saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #1: Problem after acquire image

kittidaj, 2017-04-11:
Hi Saraff
I use you example program 1 and after I pressed the scan button. The scanner scan completed but the application still show "Calibrating..." and LED on scanner blinking (Microtek ScanMaker 9800XL plus). Do you have some ideas to close this dialog box? Or I think the problem from .dll not support for calibration setting of the scanner.
Thank you very much.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-11:
Hello, kittidaj.
Try [Saraff.Twain.Sample2](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/content/Saraff.Twain.Samples.zip). but likely your problem has related with a driver of scanner.

kittidaj, 2017-04-11:
I used the sample 1 and I found 2 devices of scanner in fugure 1. There are Microtek ScanWizard Pro and WIA-A3-1600 Plus. I want to scan with FilmScanMode so I mush run through Microtek ScanWizard Pro (the program of scanner that set the parameter inside to FilmScanMode). And I got the problem follow figure2 after acquire() the program show "calibrating..." for long time. But when I selected WIA-A3-1600 Plus, It can scan completely in normal mode (reflective) that show in figure3.

figure1: https://drive.google.com/open?id=0B9ka0DDngSjmazdJLVNvYlgwRUU
figure2: https://drive.google.com/open?id=0B9ka0DDngSjmdDNHZWJPX2F3UVk
figure3: https://drive.google.com/open?id=0B9ka0DDngSjmbTA4bDVzMUNkdUk

How can I send an acknowledge or command to cancel this dialog box ("calibrating...")?
Can It change from dll?

Thank you very much for support me.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-12:
Hello, kittidaj.
Your problem has related with a driver of scanner. It want do "calibratiion". This behavior can't be changed over TWAIN protocol. I advice send question to a support service of your scanner.

closed 2017-04-17 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A