saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #2: cannot get started (it worked/no data source)

yisman, 2017-04-20:
hi
in continuation of the discussion in codeplex.. (sorry i was away for a few weeks)
i dont have a directory C:\Windows\twain_64  at all
i can make one, but what should i put into it?
thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-20:
Hello, @yisman.
Try install [Saraff.Twain.DS.Screen x64](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.DS/wiki/content/Saraff.Twain.DS.Screen.zip) and check a C:\Windows\twain_64 directory.

closed 2017-04-25 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

yisman, 2017-04-25:
hi
made some progress.
now i get some sort data source in the sample application. but pressing acquire only gives me an image of the screen. not the pic from the scanner.
please advise
thank you so much

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-25:
Hello, @yisman. 
You must install a driver of scanner that provide a TWAIN Data Source (DS) (version 64-bit). Also, you can use outproc invocations (see [samples](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/outproc)).