saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #3: Need to scan only document area.

patv2learn, 2017-04-25:
Hello Saraff,

First of all, great work. Your project helped a lot and saved lot of time and effort in my task.

I have a question and I am trying to look for different options but no success yet. My question is if I put document which is not of full scanner surface size then project still scan full letter size area and gives me with white background on empty space. Is there anyway I can just scan the document area? I am not 100% sure where the document will be on scanner and some times it may be of different size.

For example, I have Epson XP-420 scanner and I have put a business card on scanner surface. Now I want to get output where only business card should scan, no other background.

Thanks in advance
Pat.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-25:
Hello,  @patv2learn.
If you know in advance  the size of the document area, that you can use a [Twain32.ImageLayout](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Image-Layout#image-layout) property. Also, you can use a Twain32.Capabilities.SupportedSize(ICAP_SUPPORTEDSIZES) to set typical page sizes.

patv2learn, 2017-04-26:
Thanks for the detail Saraff, I will try to play with what you suggested. 

As such in my case it will be most of the time business card but the problem is user can put business card on any part of scanner surface so ideally I need to know which part of the image has content from scanned image.

I have tried few options with Magick.NET but it is not giving correct result to crop image part from full page scanned image.

Any help on this subject will be greatly appreciated. 

Thanks,
Pat.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-04-27:
Hello, @patv2learn. 
You can use capabilities of canner. Try check supporting follow capabilities on your scanner:
* `Twain32.Capabilities.AutomaticBorderDetection.IsSupported()` (ICAP_AUTOMATICBORDERDETECTION)
* `Twain32.Capabilities.AutomaticLengthDetection.IsSupported()` (ICAP_AUTOMATICLENGTHDETECTION)
* `Twain32.Capabilities.AutoSize.IsSupported()` (ICAP_AUTOSIZE)

If they no supported, you can do follow:
* apply image filter to reduce color depth (convert to B/W), calculate empty rows and discard them;
* two pass to image acquire. First pass is preview (with low image resolution and low color depth). After, user can select required area and perform second pass (with high image resolution and high color depth).

closed 2017-05-02 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A