saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #4: InvalidCast Saraff.Twain.Twain32.Enumeration to Saraff.Twain.TwSX

VitorVidal, 2017-05-03:
Hello Saraff,

I'm getting an invalid cast exception when click "save" at device app. I can see that comes from method GetCurrent() 
from TwainCapabilities.cs file. 

After some tests using sample 1, I find that method "GetCurrentCap" is returning a Twain32.Enumeration type 
and is trying to cast to Twain.TwSX type.

Do I need configure or set something?

Thank you very much.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-05-03:
Hello, @VitorVidal. 
Try send to me a Stack Trace of your exception. And, what is a "device app"?

VitorVidal, 2017-05-03:
Hello,

This is the Stack Trace that visual studio give me:

"System.InvalidCastException: Specified cast is not valid.\r\n   em Saraff.Twain.TwainCapabilities.Capability`1.GetCurrent() on C:\\Users\\vitor.vidal\\Downloads\\Saraff.Twain.NET-master\\TwainCapabilities.cs:linha 1589"

I called "device app" that app that opens when you clicked "Scan" on sample 1.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-05-03:
But in a Saraff.Twain.Sample1 don't exist line to invoke a GetCurrent method. Where you has invoked a GetCurrent?

VitorVidal, 2017-05-03:
When I first run Saraff.Twain.Sample1, I got nothing from the device when click save. So I uninstall nuget package Saraff.Twain from Saraff.Twain.Sample1 then make reference to the Saraff.Twain project that I download, so I could make a new test and got the exception.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-05-03:
It is impossible. I reviewed code of Saraff.Twain.Sample1 again. Likely you made changes of a Scan button handler code (button1_Click method). And you added invoke of a GetCurrent method at button1_Click. But your code contains error of types cast. Please, check it.

VitorVidal, 2017-05-04:
I assure you that I didn't change Saraff.Twain.Sample1 code.
Is there something that I could send to you to confirm this behaviour? 
I notice that only happens with one camera model.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-05-04:
Try execute a Saraff.Twain.Sample3 and send to me a console output.

VitorVidal, 2017-05-04:
Saraff.Twain.NET 1.0.24.579
Copyright © SARAFF 2011

ShowUI [Y/n]: Y
ShowUI = Y
IsTwain2Enable [y/N]: y
IsTwain2Enable = Y
IsTwain2Supported = Y

Select Data Source:
0: VALCam USB Twain 2.1
1: Image Capture Plus
[0]: 0
Data Source: VALCam USB Twain 2.1

Acquire Error: ReturnCode = Success; ConditionCode = Success;
TwainException: Conversão especificada não é válida.

Its portuguese, means "Invalid cast"

VitorVidal, 2017-05-04:
Sorry, have more data:

InvalidCastException: Conversão especificada não é válida.
   em Saraff.Twain.TwainCapabilities.Capability`1.GetCurrent()
   em Saraff.Twain.Twain32._TwCallbackProcCore(TwMSG msg, Action`1 endAction)
   em Saraff.Twain.Twain32._MessageFilter.PreFilterMessage(Message& m)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-05-04:
Likely, your a VALCan don't compliant a TWAIN Specification.
In a Twain32 class exist follow line of code:
```c#
switch(this.Capabilities.XferMech.GetCurrent()) {
    // ...
}
```
and your DS return invalid result (or invalid container). See bellow part of a TWAIN Specification.

# ICAP_XFERMECH
## Description
Allows the application and Source to identify which transfer mechanisms the source supports.
## Application
The current value of ICAP_XFERMECH must match the data argument type when starting the
transfer using the triplet: DG_IMAGE / DAT_IMAGExxxxXFER / MSG_GET.
## Containers
- **MSG_GET**  
	- TW_ONEVALUE (permitted TWAIN 2.1 and earlier)
	- TW_ENUMERATION (required TWAIN 2.2 and later)
- **MSG_GETCURRENT** TW_ONEVALUE   **<- THAT container must be using for a GetCurrent() method**
- **MSG_GETDEFAULT** TW_ONEVALUE
- **MSG_SET** TW_ONEVALUE
- **MSG_SETCONSTRAINT** 
	- TW_ONEVALUE
	- TW_ENUMERATION
- **MSG_RESET** TW_ONEVALUE
- **MSG_QUERYSUPPORT** TW_ONEVALUE
## Required By
All Image Sources
## TWAIN Version Introduced
Version 1.0

_10-210 TWAIN 2.4 Specification_

VitorVidal, 2017-05-04:
I'll try contact the manufacturer.
Thanks for your help.

closed 2017-05-04 by VitorVidal