saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #7: GetAllSorces throws at TwRC.EndOfList

maikschott, 2017-07-10:
Greetings,

I'm using using your TWAIN suite in outproc mode, which in general works very well, except that Saraff.Twain.Aux_MSIL.exe may quit with the Windows unhandled exception dialog which can not be suppressed by the calling application and therefore may be rather irritating for the end user.

The issue is that GetAllSorces `dsmEntry.DsmInvoke(this._AppId,TwDG.Control,TwDAT.Identity,TwMSG.GetFirst,ref _item)` (line 1437) returns TwRC.EndOfList. The following call to `new TwainException(_GetTwainStatus(); _rc)` (line 1438) then throws an ironic "It worked!" exception as _GetTwainStatus() returned TwCc.Success.

Would it be possible to include an `if (_rc == TwRc.EndOfList) return;` before the throw, so that in this case no exception in thrown, considering that in line 1448 there is a check against `_rc==TwRC.EndOfList` in which case the enumeration is stopped.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-07-10:
Hello, @maikschott.
I added requested changes. See [commit 364d9fb43b8bf5c0fbd9413fb31bcc3a859b6739 
 ](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/commit/364d9fb43b8bf5c0fbd9413fb31bcc3a859b6739).
But a `this.dsmEntry.DsmInvoke(this._AppId,TwDG.Control,TwDAT.Identity,TwMSG.GetFirst,ref _item)` (DG_CONTROL / DAT_IDENTITY / MSG_GETFIRST) can not return TwRC.EndOfList (TWRC_ENDOFLIST), because it not TWAIN compliant behavior. See 7-62 TWAIN 2.4 Specification.
# DG_CONTROL / DAT_IDENTITY / MSG_GETFIRST
## Call
`DSM_Entry(pOrigin, NULL, DG_CONTROL, DAT_IDENTITY, MSG_GETFIRST,
pSourceIdentity);`
`pSourceIdentity `= A pointer to a TW_IDENTITY structure.
## Description
The application may obtain a list of all Sources that are currently available on the system which
match the application’s supported groups (DGs, that the application specified in the
SupportedGroups field of its TW_IDENTITY structure). To obtain the complete list of all available
Sources requires invocation of a series of operations. The first operation uses MSG_GETFIRST to
find the first Source on “the list” (whichever Source the Source Manager finds first). All the
following operations use DG_CONTROL / DAT_IDENTITY / MSG_GETNEXT to get the identity
information, one at a time, of all remaining Sources.
**Note**: The application must invoke the MSG_GETFIRST operation before a MSG_GETNEXT
operation. If the MSG_GETNEXT is invoked first, the Source Manager will fail the
operation (TWRC_ENDOFLIST).
If the application wants to cause a specific Source to be opened, one whose ProductName the
application knows, it must first establish the existence of the Source using the MSG_GETFIRST/
MSG_GETNEXT operations. Once the application has verified that the Source is available, it can
request that the Source Manager open the Source using DG_CONTROL / DAT_IDENTITY /
MSG_OPENDS. The application must not execute this operation without first verifying the
existence of the Source because the results may be unpredictable.
## Return Codes
- TWRC_SUCCESS
- TWRC_FAILURE
    - TWCC_NODS /* No Sources can be found */
    - TWCC_LOWMEMORY /* Not enough memory to perform this operation*/

_7-62 TWAIN 2.4 Specification_

maikschott, 2017-07-11:
It may not be a valid behaviour, but in my case its the result I get. And it doesn't seem to be that uncommon: if you look at the DSM of Wine which is based on a Corel DSM from 2000 (according to the file header) https://github.com/wine-mirror/wine/commit/620e81a9828a6a07a449d9b7e1a8a6ecf8c02f08 for several years it did return TWRC_ENDOFLIST/TWCC_SUCCESS, before it was corrected to  TWRC_FAILURE/TWCC_NODS. My Ricoh DSM seems to do the same.

If you are uncomfortable with my proposal, which is understandable as its kind of workaround to handle non-compliant TWAIN drivers, it may also be better to at least catch TwainExceptions in Saraff.Twain.Aux_x86's and Saraff.Twain.Aux_MSIL's main method and return with an error exit code instead of not handling any exception and letting the programs just crash.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-07-11:
> May I also propose to catch TwainExceptions in Saraff.Twain.Aux_x86's and Saraff.Twain.Aux_MSIL's main method and return with an error exit code instead of just letting the programs crash.

You may add any code that you need, because Saraff.Twain.Aux_x86 and Saraff.Twain.Aux_MSIL it is samples of hosting the Saraff.Twain.Aux.

maikschott, 2017-07-12:
Thank you for your help.

closed 2017-07-12 by maikschott