saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #9: Value cannot be null. Parameter name: type

fizzycat2001, 2017-09-21:
Help, I've new in installing the web sample version of Saraff Twain.  Managed to installed the cab file.  Created a website and using the sample web pages create the index1 and index2 pages.  Everything seems to work fine except that it keeps giving the alert "Value cannot be null.  Parameter name: type".  Did I left out something.

[test.zip](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/files/1320770/test.zip)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-09-21:
Hello, @fizzycat2001.
In first. Are you tried started the Saraff.Twain.WebSample?
In second. In your sample (test.zip) missing ePTS.Twain.dll (and ePTS.Twain.ScanControl class).

fizzycat2001, 2017-09-22:
My apology, attached is the ePTS.Twain dll.  Yes I did try running the orginal WebSample but got unknown runtime error message.

[ePTS.Twain.dll.zip](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/files/1323165/ePTS.Twain.dll.zip)

fizzycat2001, 2017-09-22:
Hi again.  I had the runtime error when running on IE8 mode.  But when I switch to IE10, got the following error:  Can't load DSM, at Saraff.Twain32.OpenDSM()
      at Saraff.Twain.WebSample.ScanControl._LoadDSM()
      at Saraff.Twain.WebSample.ScanControl.Construct(ReadOnlyCpllection 1 args)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-09-22:
Hello, @fizzycat2001.
You need to install a TWAINDSM.dll (see [Installation of the Data Source Manager](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Installation-of-the-Data-Source-Manager)). If you have a 64-bit OS that you must install both version (32-bit and 64-bit) of a dll.

fizzycat2001, 2017-09-22:
Hi Saraff,
No idea how to install so I had manually copied the TWAINDSM.dll 32 bit version to the Windows/System32 folder and the 64 bit version to the Windows/SysWOW64 folder.  Is this correct?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-09-22:
Hello, @fizzycat2001 
You need to copy 64 bit version to the Windows/System32 folder and the 32 bit version to the Windows/SysWOW64 folder.

fizzycat2001, 2017-09-22:
Thanks Saraff, managed to get the sample working but there are still some glitches here and there.  Will try to iron it out on my own first.  Will need your help again if I get stuck.  Many thanks and appreciate.

closed 2017-09-22 by fizzycat2001