saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #10: Saraff.Twain.Sample3 can not execute in command line mode

nhan3bin, 2017-10-12:
Dir Mr Saraff,
Thank you so much for your project.
My issue is when I compile and execute your Saraff.Twain.Sample3, it works fine, but in command line mode, the example shows error and halt.
[image]
My platform: windows 10 64bit, Visual studio 2015 and had installed Package Saraff.Twain -Version 1.0.27.606 via NUGet Package Manager.
How can I resolve the problem?
Thank for your help.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-10-12:
Hello, @nhan3bin 
Likely, your a Data Source is not fully compliant to the TWAIN Specification. In a sample you use the Buffered Memory transfer mechanism. The Twain32._GetImageInfo method invoke a DG_IMAGE / DAT_IMAGEINFO / MSG_GET triplet. All the Data Sources must support this triples when processing image data transfer. 
You may try use the Native or Disk File (except the Buffered Memory) transfer mechanism (see [Native Mode Transfer](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Native-Mode-Transfer) and [Disk File Mode Transfer](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Disk-File-Mode-Transfer))

[image]

For resolve this problem you may contact with a customer service of manufacturer your scanner. 
Also you can see the TWAIN Specification:
## Call
`DSM_Entry(pOrigin, pDest, DG_IMAGE, DAT_IMAGEINFO, MSG_GET,pImageInfo);`
pImageInfo = A pointer to a TW_IMAGEINFO structure.
## Valid States
6 and 7 (State 7 only after receiving TWRC_XFERDONE)
## Description
When called in State 6, this operation provides to the application general image description
information about the image about to be transferred.
When called in State 7, this operation provides the Application with specific image description
information about the current image that has just been transferred. It is important during a
Memory transfer to call this triplet only after TWRC_XFERDONE is received, since that is the only
time the Source will know all the final image information.
The same data structure type is used regardless of the mechanism used to transfer the image
(Native, Disk File, or Buffered Memory transfer).
## Application
The Application can use this operation to check the parameters of the image before initiating the
transfer during State 6, or to clarify image parameters during State 7 after the transfer is complete.
Applications may inform Sources that they accept -1 value for ImageHeight/ImageWidth by
setting the ICAP_UNDEFINEDIMAGESIZE capability to TRUE.
Should the Application decide to invoke any Source features that allow the image description
information to change during scanning (such as ICAP_UNDEFINEDIMAGESIZE) and still wish to
transfer in Buffered memory mode, a DG_CONTROL/DAT_IMAGEINFO/MSG_GET call must be
made in State 7 after receiving TWRC_XFERDONE to properly interpret the image data. This is not
the default behavior of the Source.
Note that the speed at which the Application supplies buffers may determine the scanning speed.
## Source
For maximum compatibility with applications, Data Source writers are strongly encouraged to
report back finished image values in State 6. In other words, calls to DAT_IMAGEINFO should
return the same identical values in State 6, and in State 7 after TWRC_XFERDONE has been issued to
the application.
During State 6 - Fills in all fields in pImageInfo. All fields are filled in as you would expect with
the following exceptions:
- XResolution or YResolution
Set to -1 if the device creates data with no inherent resolution (such as a digital camera).
- ImageWidth
Set to -1 if the image width to be acquired is unknown (such as when using a hand-held
scanner and dragging left-to-right) , and the Application has set
ICAP_UNDEFINEDIMAGESIZE to TRUE. In this case the Source must transfer the image in
tiles.
- ImageLength
ImageLength—Set to -1 if the image length to be acquired is unknown (such as when using a
hand-held scanner and dragging top-to-bottom), and the Application has set
ICAP_UNDEFINEDIMAGESIZE to TRUE.
During State 7 - Fills in all fields in pImageInfo. All fields are filled in as during State 6, except
ImageWidth and ImageLength MUST be valid. Source shall return TWRC_SEQERROR if call is
made before TWRC_XFERDONE is sent.

_7-120 TWAIN 2.3 Specification_

nhan3bin, 2017-10-13:
> <https://maps.google.com/?q=7+(State+7&entry=gmail&source=g> only after

Dir Mr Saraff,

Thanks for your reply. I try to build the project in x86 CPU configuration
and set ScanEnableTwain2x = false.

The example works fine in command line mode.

Thanks so much.

closed 2017-10-17 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A