saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #11: Unrecognized MSG DG DAT combination

IrinaRazel, 2017-11-05:
Доброго времени суток.

Вызываю так:
` _twain.CloseDataSource();
_twain.SelectSource();
_twain.OpenDataSource();`

и выходит ошибка
Unrecognized MSG DG DAT combination

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
Добрый день.
К описанию ошибки необходимо добавить трассировку стека. Также не помешает описание обстоятельств (кода из которого осуществляется вызов и кода, предшествующего вызову).
Также Вы можете попробовать пример Saraff.Twain.Sample2.

IrinaRazel, 2017-11-05:
Нашла, в чем у Вас ошибка.
Вот тут Вы задаете Callback2, а по спецификации 2.3 надо Callback. Я поменяла и нормально пошло
 private void _RegisterCallback() {
            TwRC _rc=this._dsmEntry.DsInvoke(this._AppId,this._srcds,TwDG.Control,**TwDAT.Callback2**,TwMSG.RegisterCallback,ref _callback);
            }
        }

IrinaRazel, 2017-11-05:
Однако теперь ошибка на
 _twain.SetCap(TwCap.SupportedSizes, TwSS.A4Letter);

"Data parameter out of range."; ReturnCode = Failure; ConditionCode = BadValue	Saraff.Twain.TwainException

 

>   at Saraff.Twain.Twain32._SetCapCore(TwCapability cap, TwMSG msg) in E:\Projects\ScanHotKeyService\Saraff.Twain.NET-master\Twain32.cs:line 949
>    at Saraff.Twain.Twain32._SetCapCore(TwCap capability, TwMSG msg, Object value) in E:\Projects\ScanHotKeyService\Saraff.Twain.NET-master\Twain32.cs:line 970
>    at Saraff.Twain.Twain32.SetCap(TwCap capability, Object value) in E:\Projects\ScanHotKeyService\Saraff.Twain.NET-master\Twain32.cs:line 1012
>

IrinaRazel, 2017-11-05:
Запускаю Saraff.Twain.Sample1
Если там выставить ShowUI = false , то вылетает ошибка DG DAT MSG out of expected sequence

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
Ок. Давайте почитаем спецификацию:
## DG_CONTROL / DAT_CALLBACK / MSG_REGISTER_CALLBACK
### Description
This triplet is sent to the DSM by the Application to register the application’s entry point with the
DSM, so that the DSM can use callbacks to inform the application of events generated by the DS.
The last argument is a pointer to an initialized TW_CALLBACK structure. The TW_CALLBACK
structure should be initialized as follows:
- CallBackProc - The callback function’s entry point, used by DSM to send DAT_NULL/MSG_xxx
- RefCon - An application defined reference constant. Returned as
_pData in callback.
**Note:** Application should refrain from assigning a pointer to RefCon if they want the same
behavior in 32bit and 64bit. RefCon is not large enough to hold a pointer as 64bit.

## DG_CONTROL / DAT_CALLBACK2 / MSG_REGISTER_CALLBACK
### Description
This triplet is sent to the DSM by the Application to register the application’s entry point with the
DSM, so that the DSM can use callbacks to inform the application of events generated by the DS.
The last argument is a pointer to an initialized TW_CALLBACK2 structure. The TW_CALLBACK2
structure should be initialized as follows:
- CallBackProc - The callback function’s entry point, used by MSG_REGISTER_CALLBACK.
- RefCon - An application defined reference constant.

## Итого. В чем разница?
Разница в поле RefCon. В TW_CALLBACK оно всегда 32 разрядное (TW_UINT32  RefCon;) в то время как в TW_CALLBACK2 его разрядность зависит от разрядности процессора (TW_UINTPTR  RefCon;).
Поэтому в примечании к DG_CONTROL / DAT_CALLBACK / MSG_REGISTER_CALLBACK сказано, что необходимо избегать использования поля TW_CALLBACK.RefCon. Для устранения этой проблемы используется DG_CONTROL / DAT_CALLBACK2 / MSG_REGISTER_CALLBACK.

Поэтому Ваше утверждение `"Вот тут Вы задаете Callback2, а по спецификации 2.3 надо Callback"` неверно.

Если это действительно влияет на работу сканера, то драйвер (Data Source) Вашего сканера не соответствует TWAIN Specification 2.x.
Вы можете попробовать использовать режим TWAIN 1.x (на 64 разрядных ОС для этого используйте примеры с суффиксом _x86, например, Saraff.Twain.Sample2_x86.exe).

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
> Если там выставить ShowUI = false , то вылетает ошибка DG DAT MSG out of expected sequence

Повторюсь: 

> К описанию ошибки необходимо добавить трассировку стека.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
> Однако теперь ошибка на
_twain.SetCap(TwCap.SupportedSizes, TwSS.A4Letter);
"Data parameter out of range."; ReturnCode = Failure; ConditionCode = BadValue Saraff.Twain.TwainException

Вероятно Вы используете недопустимое значение. Попробуйте вначале получить список поддерживаемых значений используя _twain32.Capabilities.SupportedSizes.Get() (см. [
Strongly typed accessing to a capabilities](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Strongly-typed-accessing-to-a-capabilities)). А Также, проверьте поддерживается ли используемая Вами возможность (_twain32.Capabilities.SupportedSizes.IsSupported())

IrinaRazel, 2017-11-05:
к ShowUI = false

>    в Saraff.Twain.Twain32._EnableDataSource() в E:\Projects\ScanHotKeyService\Saraff.Twain.NET-master\Twain32.cs:строка 264
   в Saraff.Twain.Twain32.Acquire() в E:\Projects\ScanHotKeyService\Saraff.Twain.NET-master\Twain32.cs:строка 282
   в Saraff.Twain.Sample1.Form1.button1_Click(Object sender, EventArgs e) в E:\Projects\ScanHotKeyService\Saraff.Twain.Samples\sources\Saraff.Twain.Sample1\Form1.cs:строка 73

IrinaRazel, 2017-11-05:
> Вероятно Вы используете недопустимое значение.

Да, так и есть. А список этих значений зависит от используемого сканера?
Просто я пытаюсь оттестировать на виртуальном сканере, т.к. иных нет
http://developer.dynamsoft.com/dwt/kb/2659

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
Вероятно, используемый Вами драйвер (Data Source) сканера не может работать без отображения пользовательского интерфейса.

IrinaRazel, 2017-11-05:
> Вероятно, используемый Вами драйвер (Data Source) сканера не может работать без отображения пользовательского интерфейса.

А вот это наврятли, я сперва ставила библиотеку от asprise и она нормально отработала без интерфейса... поэтому наврятли это драйвер...

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
Попробуйте [TWAIN sample Data Source](https://sourceforge.net/projects/twain-samples/files/TWAIN%202%20Sample%20Data%20Source/) или [Saraff.Twain.DS.Screen](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.DS/wiki#samples) в качестве виртуального сканера.

IrinaRazel, 2017-11-05:
TWAIN sample Data Source оно и есть http://developer.dynamsoft.com/dwt/kb/2659
во всяком случае инсталлер пишет, что уже такое есть

IrinaRazel, 2017-11-05:
Еще в процессе сканирования выводит в консоль
Qt: Could not initialize OLE (error 80010106)
Exception при этом не выпадает

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-05:
Все примеры (образцы кода) были проверены в т.ч. с помощью TWAIN sample Data Source ([см. раздел Samples документации](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/wiki/Samples)). (С отображением UI сканера и без него, в 32 и 64 разрядных приложениях).
Поэтому, если у Вас возникли проблемы при работе примеров (образцов кода) с TWAIN sample Data Source, то вероятнее всего он (TWAIN sample Data Source) неправильно настроен и/или поврежден. (При условии отсутствия изменений в коде примеров, внесенных Вами)

IrinaRazel, 2017-11-06:
Все же есть проблемы, я запустила на другом ПК и DSM вообще не может запустить.
Я установила два виртуальных принтера, 32 и 64 битные.
Приложение TWAIN Sample APP видит только 32 битный... Все файлы dll проверила, все на своих местах... понять не могу, в чем проблема... есть идеи?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-06:
Добрый день.
Не совсем ясно, что значит `"DSM вообще не может запустить"`, причем тут `"Приложение TWAIN Sample APP" `и `"два виртуальных принтера"`.
Какие файлы Вы проверили (имя файла, версия и полный путь к файлу)?
Проверьте наличие TWAINDSM.dll версии 2.3 (32 и 64 бит) см. [Installation of the Data Source Manager](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w0200.html). А также наличие драйвера (Data Source) сканера (32 и 64 бит).

closed 2017-11-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A