saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #12: Change Language

Genarito, 2017-11-09:
Hi, I would like to know how to change my scanner's display language because I set the capability this way:
     _twain32.SetCap(TwCap.Language, TwLanguage.ENGLISH);

Or

    _twain32.SetCap(TwCap.Language, TwLanguage.SPANISH);

And still shows "ИДеТ СКаНИроВаНИе".
Thanks in advance!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-09:
Hello, @Genarito 
Try set the Twain.Language property (you may do it in the Form Designer).

Genarito, 2017-11-09:
With `_twain32.Language = TwLanguage.SPANISH;` works!
Thank you so much!

closed 2017-11-09 by Genarito

kruherson1337, 2017-11-22:
What does this property Twain32.Language do? 

Because I change it to english and I am still getting exceptions in Russian cirilic not in english.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-11-22:
Hello, @kruherson1337 
The Twain32.Language property set language for a scanner UI. Some exception messages maybe on Russian language (that throws from inside a Twain32 class).
In your case it's mean that DS isn't open. You must invoke Twain32.OpenDataSource() before.