saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #14: Convert Saraff.Twain.NET Service Samples To Service restful api

Abdullah-FAl, 2017-12-18:
Hello Saraff How are you? I hope you are doing fine

I am a web application programmer javascript - angular framework

I was trying to find a way to access the scanner in the Windows system and I was looking 

and i find you library Saraff.Twain.NET  you created
Especially this [Saraff.Twain.NET Service Samples](https://sarafftwain.codeplex.com/releases/view/620827)
I was determined to find a way to convert (Saraff.Twain.NET Service Samples) to (Saraff.Twain.NET Service [Wcf rest service](https://www.codeproject.com/Tips/1009004/WCF-RESTful-on-Windows-Service-Host))

thank God

I succeeded in connecting to GetSources()  and  return  _result.ToList()  ResponseFormat =>  json 
  
 ```
 [OperationContract]
[WebInvoke(Method = "GET",
    ResponseFormat = WebMessageFormat.Json,
    BodyStyle = WebMessageBodyStyle.Bare,
    UriTemplate = "SourceList")]
   List<Source> GetSources();
```
[image]

And also

I succeeded in connecting to GetCaps(Source source) and return _result ResponseFormat =>  json

```
        [OperationContract]
        [WebInvoke(Method = "POST",
            ResponseFormat = WebMessageFormat.Json,
            RequestFormat = WebMessageFormat.Json,
            BodyStyle = WebMessageBodyStyle.Bare,
            UriTemplate = "SourceGetCaps")]
IEnumerable<CapEnum> GetCaps(Source source);
```
[image]

///////////**************///////////////////////////
But when Acquire(Source source, IEnumerable<CapEnum> caps) throw this exception
[image]

The server encountered an error processing the request. Please see the service help page for constructing valid requests to the service. The exception message is 'InvalidOperationException: Operation is not valid due to the current state of the object. Server stack trace: at Saraff.Twain.Aux.TwainExternalProcess.AuxProcess.Execute(TwainCommand command) at Saraff.Twain.Aux.Twain32RealProxy.Invoke(IMessage msg) Exception rethrown at [0]: at System.Runtime.Remoting.Proxies.RealProxy.HandleReturnMessage(IMessage reqMsg, IMessage retMsg) at System.Runtime.Remoting.Proxies.RealProxy.PrivateInvoke(MessageData& msgData, Int32 type) at Saraff.Twain.Twain32.set_SourceIndex(Int32 value) at Saraff.Twain.Service.TwainService.<>c__DisplayClass7_0.<Acquire>b__0(Twain32 twain) in C:\Saraff.Twain.ServiceSamples\Saraff.Twain.ServiceSamples\sources\Saraff.Twain.Service\TwainService.svc.cs:line 327 at Saraff.Twain.Aux.TwainExternalProcess.Execute(String fileName, Action`1 execCallback) at Saraff.Twain.Service.TwainService.Acquire(Source source, IEnumerable`1 caps) in C:\Saraff.Twain.ServiceSamples\Saraff.Twain.ServiceSamples\sources\Saraff.Twain.Service\TwainService.svc.cs:line 308 '. See server logs for more details. The exception stack trace is:

at Saraff.Twain.Service.TwainService.Acquire(Source source, IEnumerable`1 caps) in C:\Saraff.Twain.ServiceSamples\Saraff.Twain.ServiceSamples\sources\Saraff.Twain.Service\TwainService.svc.cs:line 366 at SyncInvokeAcquire(Object , Object[] , Object[] ) at System.ServiceModel.Dispatcher.SyncMethodInvoker.Invoke(Object instance, Object[] inputs, Object[]& outputs) at System.ServiceModel.Dispatcher.DispatchOperationRuntime.InvokeBegin(MessageRpc& rpc) at System.ServiceModel.Dispatcher.ImmutableDispatchRuntime.ProcessMessage5(MessageRpc& rpc) at System.ServiceModel.Dispatcher.ImmutableDispatchRuntime.ProcessMessage11(MessageRpc& rpc) at System.ServiceModel.Dispatcher.MessageRpc.Process(Boolean isOperationContextSet)

----------------------------------------------

I hope you help me to debug this exception 

Thank you for your excellent library

and
If you need project files I can send them to you

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-12-18:
Hello, @Abdullah-FAl 
Please, see latest a samples: [Saraff.Twain.NET HTML Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html) and [Saraff.Twain.NET Service Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/service.html)

Abdullah-FAl, 2017-12-19:
Hello   @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

Thank you for your quick reply to my problem

Thanks for the [Saraff.Twain.NET Service Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/service.html) of the library

The old copy was returned to GetCaps object

The new ones When used it returned empty array

In your opinion what is the problem?
[image]

Abdullah-FAl, 2017-12-19:
is it ok to Saraff.Twain update on the latest version 
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-12-19:
Hello, @Abdullah-FAl 
If you want use the Saraff.Twain.Service as a REST Service, you must use **`http://localhost:8081/twain-web`** endpoint
Please, see follow
 [image]

Please, see a [Saraff.Twain.NET HTML Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html)

Abdullah-FAl, 2017-12-19:
After updating

Oh my God
 finally at last
The scanner works
😃😃😍😉😤😽😽🤘✌️
[image]

# Thank you for all your great work

### Do you have a way to support you?
### Or donated to you

## In order to follow your great work on your beautiful library

------------------------------------------------------
What is the method?
  Suitable for communication?
GitHub 
Email

closed 2017-12-19 by Abdullah-FAl

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2017-12-20:
Hello, @Abdullah-FAl 
You may use the paypal for donate - paypal@saraff.ru
For communication preferable use the GitHub.

EslamElmadny, 2018-04-23:
@Abdullah-FAl  Hello I'm Trying to integrate saraff twain APIs with angular project, may you help?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-04-23:
Hello, @Abdullah-FAl 
I have sample for JS [(Saraff.Twain.NET HTML Samples)](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html), but I think that it very similar code.

EslamElmadny, 2018-04-23:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
Unfor. I am facing issue to call the Apis when I installed and run the windows service
Note that the issue not appearing when i debugged the service from vs.. 
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-04-26:
Hello, @EslamElmadny .
Add to main method in the Saraff.Twain.Aux_MSIL (and Saraff.Twain.Aux_x86) try{ ... }catch{}
```C#
[STAThread]
private static void Main(string[] args) {
    try {
        using(var _twain32 = new Twain32()) {
            _twain32.IsTwain2Enable = true;
            _twain32.ShowUI = false;
            _twain32.OpenDSM();

            TwainExternalProcess.Handler(_twain32);
        }
    } catch {
    }
}
```

Abdullah-FAl, 2018-04-27:
Yes you are right @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I have added  try{ ... }catch{}

[image]

frknuzn, 2018-08-14:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
Hello, "saraff.Twain.Aux_MSIL has stopped working" has occurred error. I did what you said by this page but the error continues .
[image]

Abdullah-FAl, 2018-08-14:
Did you add try {...} catch {}  to Program.cs file And then build The project

frknuzn, 2018-08-14:
Thanks for ansver. Yes,I did add "try {...} catch {} to Program.cs file" and build project.

Abdullah-FAl, 2018-08-14:
copy thes in to Saraff.Twain.Aux_MSIL>>>   Program.cs

/* Этот файл является частью примеров использования библиотеки Saraff.Twain.NET
 * © SARAFF SOFTWARE (Кирножицкий Андрей), 2011.
 * Saraff.Twain.NET - свободная программа: вы можете перераспространять ее и/или
 * изменять ее на условиях Меньшей Стандартной общественной лицензии GNU в том виде,
 * в каком она была опубликована Фондом свободного программного обеспечения;
 * либо версии 3 лицензии, либо (по вашему выбору) любой более поздней
 * версии.
 * Saraff.Twain.NET распространяется в надежде, что она будет полезной,
 * но БЕЗО ВСЯКИХ ГАРАНТИЙ; даже без неявной гарантии ТОВАРНОГО ВИДА
 * или ПРИГОДНОСТИ ДЛЯ ОПРЕДЕЛЕННЫХ ЦЕЛЕЙ. Подробнее см. в Меньшей Стандартной
 * общественной лицензии GNU.
 * Вы должны были получить копию Меньшей Стандартной общественной лицензии GNU
 * вместе с этой программой. Если это не так, см.
 * <http://www.gnu.org/licenses/>.)
 * 
 * This file is part of samples of Saraff.Twain.NET.
 * © SARAFF SOFTWARE (Kirnazhytski Andrei), 2011.
 * Saraff.Twain.NET is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Lesser General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 * Saraff.Twain.NET is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU Lesser General Public License for more details.
 * You should have received a copy of the GNU Lesser General Public License
 * along with Saraff.Twain.NET. If not, see <http://www.gnu.org/licenses/>.
 * 
 * PLEASE SEND EMAIL TO:  twain@saraff.ru.
 */
using System;
using System.Linq;
using System.Text;
using Saraff.Twain.Aux;

namespace Saraff.Twain.Aux_MSIL {

    internal class Program {

        [STAThread]
        private static void Main(string[] args) {

            try
            {

                using (var _twain32 = new Twain32())
                {
                    _twain32.IsTwain2Enable = true;
                    _twain32.ShowUI = false;
                    _twain32.OpenDSM();

                    TwainExternalProcess.Handler(_twain32);

                   
                }

            }
            catch (Exception ex)
            {
                Console.WriteLine("Exception occured: " + ex.Message);
            }

        }

      

    }

}

frknuzn, 2018-08-14:
My Code is same.
Could the problem be related to this
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-08-14:
Hello, @frknuzn 
Has you replaced a Saraff.Twain.Aux_MSIL.exe assembly in a C:\Program Files\SARAFF\Saraff.Twain.Service directory?

frknuzn, 2018-08-15:
Hello, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I did it like you said. I changed my code and build. I replaced C:\Program Files\SARAFF\Saraff.Twain.Service 
My Project \Saraff.Twain.Aux_MSIL\bin\Release
[image]
C:\Program Files\SARAFF\Saraff.Twain.Service 
[image]
I do not get the old error anymore.
But New error occurred
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-08-15:
Hello, @frknuzn 
Likely you don't have any a TWAIN Data Source. Try install virtual scanner (e.g. [Saraff.Twain.DS.Screen](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainds/#samples))

DHIVYA-CYBERTREE, 2018-09-13:
[image]

Hi Saraff,
I am using the windows service for scanning the image. In that I am using the GetSources() method. But i am getting list thz only.Can you please help to solve thz.I added the reference also. But i didnt get the source name. I saw ur sample project also.but its not working in my windows service sample application.

DHIVYA-CYBERTREE, 2018-09-13:
Hi,
I am not getting TWAIN2 Free Software source from GetSources() method. I don't know what issue in that one. how i will get that source. I already added the saraff.Twain.DS reference also.Any one can help me solve thz. Plz

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-14:
Hello, @DHIVYA-CYBERTREE 
In first, try install a [Saraff.Twain.DS.Screen](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainds/#samples) (both x86 and x64). And after, try invoke a GetSources.

tavo87, 2020-06-19:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A

Good afternoon, can you help me please indicating
What is the correct input frame for using the next method: 

http://localhost: 8081/twain-web/Acquire

if it is possible to deliver plot or example from postman.
Thank you

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-06-19:
Hello, @tavo87 
```JavaScript
            $.getJSON(
                _serviceUrl + "Acquire",
                {
                    platform: _platforms[_ds.Platform],
                    version: _versions[_ds.Version],
                    sourceId: _ds.Id,
                    caps: JSON.stringify({
                        IPixelType: $("#PixelType").val(),
                        XResolution: $("#dpi").val(),
                        YResolution: $("#dpi").val()
                    })
                },
                function (_image) {
                    $("#notify").text("Ready.");
                    $("#image").prop("src", "data:image/jpeg;base64," + _image);
                    $("#image").data({ base64: _image });
                }).fail(function () {
                    $("#notify").text("An error occurred during Acquire.");
                }).always(function () {
                    $("#toolstrip").children().prop("disabled", false);
                });
```
See [Saraff.Twain.NET HTML Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html)

tavo87, 2020-06-19:
Thank you
[image]