saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #18: Detect Flatbed

MrBlueHere, 2018-02-06:
First of all, thank you for great effort put into this project, really appreciate it.

I would like to ask: is there a way to find out if scanner has a flatbed? Something similiar to: this._twain32.IsCapSupported(TwCap.FeederEnabled) but Flatbed instead of Feeder.

Thank you.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-02-07:
Hello, @MrBlueHere 
Thank for you feedback.
I reviewed all a capabilities but I don't find anywhere that can be detect  flatbed on scanner. I think it because that flatbed don't need to additional settings for scanning. Also, you can see list of all a capabilities in [TWAIN Specification](http://www.twain.org/specification/) (Capabilities in Categories of Functionality, TWAIN 2.4 Specification, 10-3).

MrBlueHere, 2018-02-07:
Thank you. And is there any workaround to detect, if scanner has flatbed?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-02-07:
Probably not.
Also, you can try ask this question on the twain.org  forum.

MrBlueHere, 2018-02-07:
I'll give it a try, thank you.

MrBlueHere, 2018-02-07:
Would something like this be possible in Saraff.Twain.NET?

http://www.atalasoft.com/KB/article.aspx?id=10379

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-02-07:
Yes, all a capabilities that uses in the topic is available in Saraff.Twain.NET.

closed 2018-02-15 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A