saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #20: Issue

kailash74, 2018-03-19:
[image]

using Saraff.Twain.NET HTML  we get every time this error . is there any thing to do in the environment settings. the service is running.
once we say continue it get the source from the client side . 
every time we call this error pops up any thing we are missing

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-19:
Hello, @kailash74 
Try install TWAINDSM.DLL (see [Installation of the Data Source Manager](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w0200.html)).

kailash74, 2018-03-19:
thanks will check and come back

kailash74, 2018-03-19:
hi ..

we are getting the same error

https://github.com/twain/twain-dsm/tree/master/Releases/dsm_020400/windows/64

copied and placed the dll in syswow64

but still
error persist

windows 10 system
windows  7 system

please guide

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-19:
https://github.com/twain/twain-dsm/tree/master/Releases/dsm_020400/windows/32 to syswow64
https://github.com/twain/twain-dsm/tree/master/Releases/dsm_020400/windows/64 to system32

kailash74, 2018-03-19:
> https://github.com/twain/twain-dsm/tree/master/
> Releases/dsm_020400/windows/32 to syswow64
> https://github.com/twain/twain-dsm/tree/master/
> Releases/dsm_020400/windows/64 to system32

did not get u .

i am not able to understand .

32 to be system32
64 to syswow64.

this is what i added . but still it happens

kailash74, 2018-03-20:
> Hi
>
> Please find the detail screenshot.
>
> the error persists even after copying and pasting the dlls in system
> folders.
> in windows 10 and in Windows 7    64 bit version.
>
>
>
>
>
>
>
>
>
> On Mon, Mar 19, 2018 at 7:00 PM, Kailash Krishnan <kailas74@gmail.com>

Hi

Any more detail I can get . Pls help

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-20:
Hello, @kailash74 
Try run the Saraff.Twain.Sample2.exe and Saraff.Twain.Sample_x86.exe from [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html). It can help understand your problem.

kailash74, 2018-03-20:
Hi.

The Error is Occuring in Win 10 system only Now.

we tried the same .net core application in win 7 no issue but in windows 10
still the error persist. using jtwain.js

could not really resolve what is the issue related to DSM .

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-20:
Are you tried run the Saraff.Twain.Sample2.exe and Saraff.Twain.Sample_x86.exe on windows 10 ?

kailash74, 2018-03-20:
Hi

Saraff.twain.sample_x86.exe result in Windows 10 its able to get the source

2.  Saraff.Twain.Sample2.exe

say cannot opendsm message dialog

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-20:
Try copy https://github.com/twain/twain-dsm/blob/master/Releases/dsm_020402/windows/64/TWAINDSM.dll file to c:\windows\system32 folder (if you have windows 10 64-bit). And after, try run Saraff.Twain.Sample2.exe

kailash74, 2018-03-20:
> Releases/dsm_020402/windows/64/TWAINDSM.dll file to c:\windows\system32

Please find the screen shot..

Not source getting detected

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-03-20:
Likely, DSM was loaded, but you don't have any 64-bit DS (scanners). Try install virtual scanner https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainds/download/Saraff.Twain.DS.Screen.zip

kailash74, 2018-03-20:
> 5a.github.io/sarafftwainds/download/Saraff.Twain.DS.Screen.zip

Yes now this error has gone has gone from my application too as the default
ds for  64 is getting success.

will put in a new fresh testing system and come back.

really appreciate your time . bless u