saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #21: Check if device is connected before opening DataSource

MrBlueHere, 2018-05-03:
Hi,

I ran into a little issue. I have a code set up, where I loop through twain.SourcesCount and retrieve info about sources, like id, name, manufacturer, etc.  Problem arises when I call to twain.OpenDataSource(), which causes my app to freeze for some time. 
What happens is, that scanner driver will open window with info saying, that device is disconnected (this takes 10 seconds or more). Exception (DG DAT MSG out of expected sequence) is thrown only after the window is closed by the user. 
My app is being used like back-end webservice, so it‘s uncomfotable for front-end to wait for my app when it gets to this „waiting for drivers ui state“.

Also I know, that simply connecting scanner will solve the issue, I'm just trying to prevent this, so my app doesn't hang for such a long time.

Is there a way to check if a specific device is connected before opening DataSource? Or somehow block this driver ui?

By the ways, I always set Twain.Ui to false, so this doesn't help to block driver ui.

Thank you.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-05-04:
Hello, @MrBlueHere 
Unfortunately, impossible to check if a specific device is connected before opening DataSource. But you can use background thread for opening a DS in external process. See the Saraff.Twain.OutprocSample4 in the [Saraff.Twain.NET Outproc Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/outproc.html).

MrBlueHere, 2018-05-04:
Thank you, outproc may help.

closed 2018-05-10 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A