saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #22: Saraff.Twain.NET in html and feeder scanners

rahimzade, 2018-06-24:
hi 
in features of Saraff.Twain.NET you write : Control the work of Automatic Document Feeder (ADF) while scanning
i tested Saraff.Twain.NET HTML Samples from this [link](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html)
this work fine for scan one page. but in need to scan multipage in html. can you help me for this one? i need to document for this .
thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-06-24:
Hello, @rahimzade
In JS script try set FeederEnabled capability to a true (if your scanner it support).
For more info see (accessing to a capabilities)[https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2110.html]

rahimzade, 2018-06-25:
hi 
i try to set capability  true, but cant set this :(
can you introduce a sample or article by detail for this?
thanks

rahimzade, 2018-06-26:
yes
i can write this code

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-06-26:
Hello, @rahimzade
It is nice. But may be you show it.

rahimzade, 2018-06-27:
hi saraff
in sample script i change code The following this :
`  $("#acquire").click(function () {
            $("#toolstrip").children().prop("disabled", true);
            $("#notify").text("Acquire...");

            let _ds = JSON.parse($("#ds").val());
            $.getJSON(
                _serviceUrl + "Acquire",
                {
                    platform: _platforms[_ds.Platform],
                    version: _versions[_ds.Version],
                    sourceId: _ds.Id,
                    caps: JSON.stringify({
                        IPixelType: $("#PixelType").val(),
                        XResolution: $("#dpi").val(),
                        YResolution: $("#dpi").val(),
                        DuplexEnabled: true
                    })
                },
                function (_image) {
                    $("#notify").text("Ready.");
                    $("#image").prop("src", "data:image/jpeg;base64," + _image);
                    $("#image").data({ base64: _image });
                }).fail(function () {
                    $("#notify").text("An error occurred during Acquire.");
                }).always(function () {
                    $("#toolstrip").children().prop("disabled", false);
                });
        });`
i added : `DuplexEnabled: true` 
but in scan i get error
can you help me?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-06-27:
Hello, @rahimzade
"DuplexEnabled: true" is not valid. You must request supported values from wcf-service, like as XResolution.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-07-02:
Hello, @rahimzade.
You can see below sample of code that request supported values of the FeederEnabled capability from wcf-service.

```javascript
        $("#ds").change(function () {
            // ...
            let _ds = JSON.parse($(this).val());
            $.getJSON(
                _serviceUrl + "GetCaps",
                { platform: _platforms[_ds.Platform], version: _versions[_ds.Version], sourceId: _ds.Id, caps: "IPixelType,XResolution,FeederEnabled" /* + all a capabilities that you need */ },
                function (_caps) {
                    // ...
                }).fail(function () {
                    $("#notify").text("An error occurred during Get Capabilities.");
                }).always(function () {
                    $("#toolstrip").children().prop("disabled", false);
                });
        }).change();
```

closed 2018-07-06 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A