saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #23: Saraff.Twain.Sample3 has error : specified cast is not valid

parinazk, 2018-07-05:
Dir Mr Saraff,
My issue is when I select resolution in your Saraff.Twain.Sample3, shows error specified cast is not valid.

[image]

How can I resolve the problem?
Thank for your help.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-07-05:
Hello, @parinazk 
Likely, your a data source (scanner driver) is not compliant TWAIN Specification. Try use another a data source.

parinazk, 2018-07-07:
I try another data source (WIA) in Saraff.Twain.Sample3,could scan but Saraff.Twain.Sample2 could not scan by WIA driver and show error:
TwainException: DG DAT MSG out of expected sequence. ConditionCode = SeqError; ReturnCode = Failure;
How can I resolve the problem?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-07-07:
Hello, @parinazk 
Problem in your a data source (scanner driver), it is not compliant TWAIN Specification. Try use another a data source. You can write about this problem to a custom service of HP. Also try update your driver of scanner to latest version.

closed 2018-07-11 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A