saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #25: Multiscan Page

frknuzn, 2018-09-24:
I'd like multiple scan but it is not multiple don't scan

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-24:
Hello, @frknuzn 
Likely a issue #22 may help to you.

frknuzn, 2018-09-25:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I read this issue #22 and applied. But my problem is not solved. I using Saraff.Twain.DS.Screen. Is the problem Saraff.Twain.DS.Screen ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-25:
Hello, @frknuzn 
Then describe what you mean about a multiscan: that acquire same page many times or acquire many pages in one time? And what you changed in a Saraff.Twain.HtmlSamples?

frknuzn, 2018-09-25:
Hello, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I'd like acquire many pages in one time. My jtwain.js content 

`serviceUrl + "GetCaps",
                { platform: _platforms[_ds.Platform], version: _versions[_ds.Version], sourceId: _ds.Id, caps: "IPixelType,XResolution,**FeederEnabled**" },
                function (_caps) {
                    $("#notify").text("Ready.");
                    [$("#PixelType"), $("#dpi")].forEach(function (_combo) { _combo.empty(); });
                    let _notSupportedCaps = ["IPixelType", "XResolution"].filter(function (_cap) {
                        return !_caps.some(function (x) {
                            return _capabilities[x.Cap] === _cap;
                        });
                    });`

frknuzn, 2018-09-25:
How can i do XferCount ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-25:
```javascript
{ platform: _platforms[_ds.Platform], version: _versions[_ds.Version], sourceId: _ds.Id, caps: "IPixelType,XResolution,XferCount" /* + all a capabilities that you need */ },
```
And see 10-112 [TWAIN 2.4 Specification](https://www.twain.org/specification/):
### CAP_XFERCOUNT
#### Description
The application is willing to accept this number of images.
#### Application
Set this capability to the number of images you are willing to transfer per session. Common
values are:
- 1 Application wishes to transfer only one image this session
- 2 Application wishes to transfer only two image this session
- -1 Application is willing to transfer multiple images
#### Values
Type: TW_INT16
Value after MSG_OPENDS: -1
After MSG_RESET/MSG_RESETALL: -1
Allowed Values: -1 **and 1** – **32767**
#### Containers
MSG_GET 
- TW_ONEVALUE
- TW_RANGE // 2.3 and higher

MSG_GETCURRENT 
- TW_ONEVALUE

MSG_GETDEFAULT 
- TW_ONEVALUE

MSG_SET 
- TW_ONEVALUE

frknuzn, 2018-09-26:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A ,
I used your code. But it don't work. I repeat,
1. I using Saraff.Twain.HtmlSamples.
2. Due to this error " https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/14#issuecomment-412840543 " **I using Saraff.Twain.DS.Screen**
3. I did not make any changes except jtwain.js page

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
Hello, @frknuzn 
I understand you. But, if you wanna make application that use a feeder of scanner, you must use scanner with support a feeder. The Saraff.Twain.DS.Screen don't support feeder (because it make image from screen, and  feeder is missing in screen).
If you wanna use a Saraff.Twain.DS.Screen and acquire many images per session than you may use a CAP_XFERCOUNT capability.
The Saraff.Twain.HtmlSamples it is sample of using a Saraff.Twain.NET, but it not commerce application.

frknuzn, 2018-09-26:
Hello again @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I couldn't solve the error " #14 (comment) " . So I used saraf Saraff.Twain.DS.Screen. I think, I made a mistake in the packaging process after making changes in Saraff.Twain.Aux_MSIL. Could you help me solve this error ? I don't use for commerce project.

frknuzn, 2018-09-26:
I added  "try catch" block but  Does not working when I pack the application. How I can packaging ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
Likely you must rebuild a Saraff.Twain.ServiceInstaller_x64 and a Saraff.Twain.ServiceInstaller_x86

frknuzn, 2018-09-26:
I added "try-catch" and rebuild setup , solved this problem https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/14#issuecomment-412840543
I used this comment content https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/22#issuecomment-401642889 but don't worked

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
Make sure that you use same version of a Saraff.Twain.NET in all projects (assemblies).

frknuzn, 2018-09-26:
Should I do this https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2110.html

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
What scanner you use? Is scanner a Saraff Screen DS (Saraff.Twain.DS.Screen) or another?

frknuzn, 2018-09-26:
My scanner is supported multi-scan. I used Saraff.Twain.NET

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
You may try use follow code, but it impossible using directly from JS.
```C#
//Feeder & Duplex
if(this._twain32.Capabilities.Duplex.IsSupported(TwQC.GetCurrent)) {
    if(this._twain32.Capabilities.Duplex.GetCurrent()!=TwDX.None) {
        if(this._twain32.Capabilities.FeederEnabled.IsSupported(TwQC.Set)) {
            this._twain32.Capabilities.FeederEnabled.Set(true);

            if(this._twain32.Capabilities.DuplexEnabled.IsSupported(TwQC.Set)) {
                this._twain32.Capabilities.DuplexEnabled.Set(true);
            }
            this._twain32.Capabilities.XferCount.Set(-1);
        }
    }
}
```

frknuzn, 2018-09-26:
Where i going to this code https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25#issuecomment-424643978

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
Likely that in Saraff.Twain.Service. Or you can use follow code
```javascript
        $("#ds").change(function () {
            // ...
            let _ds = JSON.parse($(this).val());
            $.getJSON(
                _serviceUrl + "GetCaps",
                { platform: _platforms[_ds.Platform], version: _versions[_ds.Version], sourceId: _ds.Id, caps: "IPixelType,XResolution,FeederEnabled" /* <<< it is list of a capabilities that a values of which you need. you must handle returned values*/ },
                function (_caps) {
                    // ...
                }).fail(function () {
                    $("#notify").text("An error occurred during Get Capabilities.");
                }).always(function () {
                    $("#toolstrip").children().prop("disabled", false);
                });
        }).change();
```

but, it is necessary not only to add a line "FeederEnabled", but also you need to process the received values. and use them when sending a request for scanning. Likely as IPixelType and XResolution capability values.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
You can use jtwain.js as sample.
```javascript
/* adding a values of IPixelType and XResolution to comboboxes */
                    $.each(_caps, function (i, _cap) {
                        let _combo = { IPixelType: $("#PixelType"), XResolution: $("#dpi") }[_capabilities[_cap.Cap]];
                        if (_combo !== undefined) {
                            let _current;
                            $.each(_cap.Values, function (i, _val) {
                                let _item;
                                _combo.append(_item = $("<option/>").prop("value", JSON.stringify(_val.RawValue)).text(_capValues[_capabilities[_cap.Cap]](_val.Name)));
                                if (_cap.Current.Name===_val.Name) {
                                    _current = _item;
                                }
                            });
                            _combo.val(_current.prop("value"));
                        }
                    });
```

```javascript
/* scanning with a values of IPixelType and XResolution that user selected */
            let _ds = JSON.parse($("#ds").val());
            $.getJSON(
                _serviceUrl + "Acquire",
                {
                    platform: _platforms[_ds.Platform],
                    version: _versions[_ds.Version],
                    sourceId: _ds.Id,
                    caps: JSON.stringify({
                        IPixelType: $("#PixelType").val(),
                        XResolution: $("#dpi").val(),
                        YResolution: $("#dpi").val()
                    })
                },
                function (_image) {
                    $("#notify").text("Ready.");
                    $("#image").prop("src", "data:image/jpeg;base64," + _image);
                    $("#image").data({ base64: _image });
```

frknuzn, 2018-09-26:
My jtwain.js codes this same https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25#issuecomment-424668295
My code https://codeshare.io/5Q69LQ

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
And you need to add a FeederEnabled processing:
```javascript
                        let _combo = { IPixelType: $("#PixelType"), XResolution: $("#dpi"), /*FeederEnabled: $(#feeder) <<< as sample*/ }[_capabilities[_cap.Cap]];
```
```javascript
                    caps: JSON.stringify({
                        IPixelType: $("#PixelType").val(),
                        XResolution: $("#dpi").val(),
                        YResolution: $("#dpi").val(),
                        /* FeederEnabled: $("#feeder").val()  <<< as sample*/
                    })

```

frknuzn, 2018-09-26:
I wrote https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25#issuecomment-424678681 but not working.
My new code is https://codeshare.io/5Q69LQ

frknuzn, 2018-09-26:
Do i need that ? https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25#issuecomment-424643978

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-26:
in html (as sample)
```html
      <select id="ds" name="D1" style="width:300px; margin:2px;">
        <option>---</option>
      </select>
      <select id="dpi" style="width:60px; margin:2px;">
        <option>---</option>
      </select>
      <select id="PixelType" style="width:75px; margin:2px;">
        <option>---</option>
      </select>

      <select id="feeder" style="width:75px; margin:2px;">
        <option>---</option>
      </select>
```

frknuzn, 2018-09-26:
I applied 
[image]

frknuzn, 2018-09-27:
Hello @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
I did make select box selected "true" but scanner is not work multi scan mode
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-27:
Hello, @frknuzn 
Also, you need to add changes to a Saraff.Twain.Service (in TwainService.cs)
```c#
        public Stream Acquire(Source source,IEnumerable<CapEnum> caps) {
            try {
                // ...
                var _result=new MemoryStream();
                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain => {
                        twain.EndXfer+=(sender,e) => {
                            try {
                                using(var _image = e.Image) {
                                    _image.Save(_result,ImageFormat.Jpeg); // <<< you must change these lines
                                    _result.Seek(0,SeekOrigin.Begin);      // to return multiple images. you can try to make multipage tiff file and return it
                                }
                            } catch {
                            }
                        };
                        // ...
                        twain.Capabilities.XferCount.Set(1); // <<< comment this line
                        // ...
                    });
                return _result;
            } catch(Exception ex) {
                // ...
            }
        }

```

frknuzn, 2018-09-27:
I changed twain.Capabilities.XferCount.Set(-1); and multi scan is worked . But I can not return multiple image in html.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-27:
```c#
_image.Save(_result,ImageFormat.Jpeg); // <<< you must change these lines
_result.Seek(0,SeekOrigin.Begin);      // to return multiple images. you can try to make multipage tiff file and return it
```

frknuzn, 2018-09-27:
I could not. Can you share a sample code ?

frknuzn, 2018-09-28:
I tried many methods but I couldn't. Can you help me please ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-29:
Hello, @frknuzn 
You can try follow:
In a Saraff.Twain.Service:
```C#
        public Stream Acquire(Source source,IEnumerable<CapEnum> caps) {
            try {
                var _result=new MemoryStream();
                var _images = new List<System.Drawing.Image>();
                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain => {
                        twain.EndXfer+=(sender,e) => _images.Add(e.Image);
                        // ...
                    });

                var _params = new EncoderParameters(2);
                _params.Param[0] = new EncoderParameter(System.Drawing.Imaging.Encoder.Compression, (long)EncoderValue.CompressionNone);
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.MultiFrame);

                _images[0].Save(
                    _result,
                    ImageCodecInfo.GetImageEncoders().First(x => x.MimeType == "image/tiff"),
                    _params);

                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.FrameDimensionPage);

                for(var i = 1; i < _images.Count; i++) {
                    _images[0].SaveAdd(_images[i], _params);
                }
                foreach(var _image in _images) {
                    _image.Dispose();
                }

                _result.Seek(0, SeekOrigin.Begin);
                return _result;
            } catch(Exception ex) {
                // ...
            }
        }
```
In a WebSite/Scripts/jtwain.js
```javascript
                function (_image) {
                    // ...
                    $("#image").prop("src", "data:image/tiff;base64," + _image);
                    // ...
                }
```

frknuzn, 2018-10-01:
Hello @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
Thank you for answer. I applied your code .But I don't see multiple image in html.
My Saraff.Twain.Service:
public Stream Acquire(Source source,IEnumerable<CapEnum> caps) {
            try {
                var _formatter=new BinaryFormatter();
                
                var _result=new MemoryStream();
                var _images = new List<System.Drawing.Image>();
                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain => {
                        twain.EndXfer += (sender, e) => _images.Add(e.Image);

                        twain.SourceIndex=source.Id;
                        twain.OpenDataSource();

                        foreach(var _cap in caps) {
                            using(var _stream=new MemoryStream(_cap.Current.RawValue)) {
                                twain.SetCap(_cap.Cap,_formatter.Deserialize(_stream));
                            }
                        }

                        twain.Capabilities.XferCount.Set(-1);
                        twain.Capabilities.Indicators.Set(false);
                        

                        twain.Acquire();
                    });
                var _params = new EncoderParameters(2);
                _params.Param[0] = new EncoderParameter(System.Drawing.Imaging.Encoder.Compression, (long)EncoderValue.CompressionNone);
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.MultiFrame);

                _images[0].Save(
                    _result, ImageCodecInfo.GetImageEncoders().First(x => x.MimeType == "image/tiff"), _params);

                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.FrameDimensionPage);

                for (int i = 1; i < _images.Count; i++)
                {
                    _images[0].SaveAdd(_images[i], _params);
                }
                foreach (var _image in _images)
                {
                    _image.Dispose();
                }

                _result.Seek(0, SeekOrigin.Begin);
                return _result;

                
                
            } catch(Exception ex) {
                var _message=string.Empty;
                for(var _ex=ex; _ex!=null; _ex=_ex.InnerException) {
                    _message+=string.Format("{0}: {1}{3}{2}{3}{3}",_ex.GetType().Name,_ex.Message,_ex.StackTrace,Environment.NewLine);
                }
                throw new FaultException(_message);
            }
        }

My jtwain.js
$("#image").prop("src", "data:image/tiff;base64," + _image);

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-01:
Hello, @frknuzn 
In HTML you must self implement showing one of frames from multi-page tiff file.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-01:
No, I do not have such an opportunity.

frknuzn, 2018-10-01:
I don't understand this comment https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25#issuecomment-425805063
How can i do it ?

makingbloke, 2018-10-01:
I’ve done this several times (all professionally so I don’t own the copyright and can’t give the source). TIFF images are not natively supported by browsers so what you need to do in your viewer call a server side script that requests a page. The server side script then extracts the page from the TIFF and converts it into a format the browser supports such as a PNG and returns this to the client. Where it gets difficult is with black and white images. In TIFF viewers these are generally anti-aliased (edges smoothed) to make them more readable. I converted my TIFF page to a higher resolution than the screen (say 300dpi at least) this way the user can zoom without keep calling the server and anti-aliased the image on the server. It worked very well. This is all quite complex so not something you can really get a simple example for. The other easier option is to convert the TIFF to a PDF server side and then download this to your client.

frknuzn, 2018-10-01:
I show single Tiff file . I want show multiple file in html.

makingbloke, 2018-10-01:
That’s my point you can’t. You need something server side to deliver a single page.

frknuzn, 2018-10-01:
Server side is ok. I want show multiple file in html.

makingbloke, 2018-10-01:
You need a server side script that the client can call which will extract a page from the TIFF and return it to the client. Unless you’ve got a plugin or something similar most browsers don’t support multi page tiffs. You need to implement your own viewer or use a product such as ViewOne. I understand what you are trying to do.

frknuzn, 2018-10-01:
Can you share example ?

frknuzn, 2018-10-01:
I used this https://github.com/seikichi/tiff.js project and show in html. But single tiff seems

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-01:
You can upload your multi-page tiff file to a server (remote server) directly after acquire a image, and request from a server count of pages of a tiff file and a page (as jpeg file, as sample) from tiff file that your need.

frknuzn, 2018-10-02:
Hello,
I uploaded the tiff to server and seen as multiple. Thank you so much for everything.

closed 2018-10-02 by frknuzn