saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #26: Possible memory leak

bkraul, 2018-09-27:
When using Saraff.Twain component, during the AcquireCompleted event I am using:
```
var stream = new System.IO.MemoryStream();
mySaraffTwain.GetImage(i).Save(stream, ImageFormat.Tiff);
``` 
And then proceeding to save the stream to a file. I am disposing the stream, but the memory used in the twain images does not seem to be disposed, the memory usage keeps climbing, and it is not disposed even after closing the form.

I have tried using the ```.Dispose()``` method pf the component, to no avail. Is there somethin I need to do to cause memory used by the Twain component to be released?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-09-27:
Hello, @bkraul 
You must self dispose all images that you receive using a GetImage method.
As sample:
```c#
var _stream = new MemoryStream();
using(var _image = _twain.GetImage(i)) {
    _image.Save(stream, ImageFormat.Tiff);
}
```
Also, you can process a EndXfer event, as sample:
```c#
var _stream = new MemoryStream();
using(var _image = e.Image) {
    _image.Save(_stream, ImageFormat.Tiff);
}
```

bkraul, 2018-09-27:
Thanks @saraff-9EB1047A4BEB4cef8506B29BA325BD5A. That seemed to alleviate the problem. I am marking this as resolved.

closed 2018-09-27 by bkraul