saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #27: DG DAT MSG out of expected sequence when opening DataSource

MrBlueHere, 2018-10-18:
Hi,

I'm doing compatibility testing of my app with numerous scanners. So far fujitsu and brother scanners seem to work fine, but I'm having issues with Epson.

When I try to open the data source (through TwainExternalProcess), I get this error:
`Server stack trace: 
	at Saraff.Twain.Twain32.OpenDataSource()
Exception rethrown at [0]:  
	at System.Runtime.Remoting.Proxies.RealProxy.HandleReturnMessage(IMessage reqMsg, IMessage retMsg)
	at System.Runtime.Remoting.Proxies.RealProxy.PrivateInvoke(MessageData& msgData, Int32 type)
	at Saraff.Twain.Twain32.OpenDataSource()
	at ScanApp.Models.Scanner.<>c__DisplayClass63_0.<FindIfTwainDeviceOnline>b__1(Twain32 _twain, IDisposable host) 

	in D:\\Projects\\ScanApp\\ScanApp_\\ScanApp\\Models\\Scanner.cs:line 307
	at Saraff.Twain.Aux.TwainExternalProcess.Execute(String fileName, Action`2 execCallback)
	at ScanApp.Models.Scanner.<>c__DisplayClass63_0.<FindIfTwainDeviceOnline>b__0() 
	
	in D:\\Projects\\ScanApp\\ScanApp_\\ScanApp\\Models\\Scanner.cs:line 301\r\n   
	at System.Threading.Tasks.Task.InnerInvoke()
	at System.Threading.Tasks.Task.Execute()
`
More info about the error: Return Code: Failure; ConditionCode: SeqError, HResult: -2146233088

At this point I'm starting to think that the issue is because of bad drivers, but I'm not sure.
If you need more info about the error, please let me know.

Thank you very much and have a nice day.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-18:
Hello, @MrBlueHere 
You right, this problem related to scanner driver (DS). But sure that you use latest version of a TWAINDSM.dll (must by 2.3 or later), because some drivers can be install more oldest version.

MrBlueHere, 2018-10-19:
Thank you for an answer. I investigated it yeasterday into more detail and found out that it wasn't the first exception that was thrown. I will leave the info about the problem here, if someone will have the same problem as me.

Basically I was starting a new TwainExternalProcess and then opening the DataSource(DS), to check if it is connected, but I wasn't calling twain.CloseDataSource() when I finished my work. 
Now, this is a mistake on my part, the DS probably should have been released, but this was the first time it actually caused any problems (it worked alright on other scanners), so maybe Epson drivers are partly responsible for this.

My app also supports WIA and I can confirm that it was the issue also by the logs that WIA gave. It said ther "part of the file has been locked by other process" by which I understand, it was my TwainExternalProcess.

Fun thing is, that I disposed the process, so it wasn't really running, but still, my mistake. So always Close the DataSource. 

Sorry @saraff-9EB1047A4BEB4cef8506B29BA325BD5A  for the confusion.

MrBlueHere, 2018-10-19:
I'll closed this as resolved.

closed 2018-10-19 by MrBlueHere