saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #28: Detecting duplex (and other capabilites), the right way

MrBlueHere, 2018-10-19:
Hi,

me again, I'm sorry if I'm bothering. I was testing a lot of scanners, so I have lot of data.
Overall, I can confirm that your library works great, the issues I encountered where mostly caused by me or incompatible drivers. So thank you very much.

My question:
How exactly do you verify that a capability is supported for sure? In particular DuplexEnabled capability.
My current code for checking sources(by sources i mean feeder, duplex, ...):
`
 foreach (var _sc in sources)
                    {
                        if ((twain.IsCapSupported(_sc) & TwQC.GetCurrent) != 0)
                        {
                            if (_sc == TwCap.FeederEnabled)
                            {
                                if ((twain.IsCapSupported(TwCap.FeederEnabled) & TwQC.Get) != 0)
                                {
                                    feederPresent = true;
                                }
                            }

                            else if (_sc == TwCap.Duplex)
                            {
                                if ((twain.IsCapSupported(TwCap.DuplexEnabled) & TwQC.Get) != 0)
                                {
                                    duplexPresent = true;
                                }
                            }
                        }
                    }
`
Is it enough to just check `((twain.IsCapSupported(_sc) & TwQC.GetCurrent) != 0)` to get the current value of a capability and verify that it is in fact supported, or do I also need to do `((twain.IsCapSupported(_sc) & TwQC.Get) != 0)` ?

I was testing Plustek PL3060 and by using this method it didn't detect its duplex, but I can verify that duplex is in fact supported, if I do `twain.SetCap(TwCap.DuplexEnabled, true);` it scans with duplex just fine.

Thanks and have a nice day.

MrBlueHere, 2018-10-22:
Okay, thanks, so I was thinking about doing something like this:

```
// Check duplex
                    if (twain.Capabilities.DuplexEnabled.IsSupported(TwQC.GetCurrent) && twain.Capabilities.DuplexEnabled.GetCurrent() == true)
                    {
                        duplexPresent = true;
                    }
                    else if (twain.Capabilities.DuplexEnabled.IsSupported(TwQC.Get))
                    {
                        var _all = twain.Capabilities.DuplexEnabled.Get();

                        for (var i = 0; i < _all.Count; i++)
                        {
                            if (_all[i].Equals(true))
                            {
                                duplexPresent = true;
                            }
                        }
```
And the same process for fedder, but with twain cabality FeederEnabled

Does it seem okay?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-22:
Ok. But better invoke immediately a Get, it's faster.
So, you can check only a Get operation.
```C#
if(this._twain.Capabilities.DuplexEnabled.IsSupported(TwQC.Get)) {
    // a capability is supported a TwQC.Get operation, but it may be supported not all values (only a false, as sample)
    var _all = this._twain.Capabilities.DuplexEnabled.Get(); // get all supported values
    for(var i = 0; i < _all.Count; i++) { // you can check supported a true, or not
        if(_all[i].Equals(true)) {
            // <<< Duplex Supported
        }
    }
    // ...
}
```

MrBlueHere, 2018-10-22:
Okay, I see, thanks. I'll close the issue as resolved.

closed 2018-10-22 by MrBlueHere