saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #29: Memory not getting released soon enough

MrBlueHere, 2018-10-22:
Hi,

When doing scan operation, memory climbs a lot (it eventually is released tho), but it can cause problems when scanning at high resolution and feeder. What I see happenning is that the memory usage climbs after every new image being processed and thank, when there is not enough continous memory available, if fails with: 

Saraff.Twain.TwainException: Exception of type 'System.OutOfMemoryException' was thrown. ---> System.OutOfMemoryException: Exception of type 'System.OutOfMemoryException' was thrown.
   at System.IO.MemoryStream.set_Capacity(Int32 value)
   at System.IO.MemoryStream.EnsureCapacity(Int32 value)
   at System.IO.MemoryStream.Write(Byte[] buffer, Int32 offset, Int32 count)
   at System.IO.BinaryWriter.Write(Byte[] buffer, Int32 index, Int32 count)
   at Saraff.Twain._ImageHandler.PtrToStreamCore(IntPtr ptr, Stream stream)
   at Saraff.Twain.DibToImage.PtrToStreamCore(IntPtr ptr, Stream stream)
   at Saraff.Twain._ImageHandler.PtrToStream(IntPtr ptr, IStreamProvider provider)
   at Saraff.Twain.Twain32._NativeTransferPictures()
   at Saraff.Twain.Twain32._TwCallbackProcCore(TwMSG msg, Action`1 endAction)
   at Saraff.Twain.Twain32._MessageFilter.PreFilterMessage(Message& m)
   --- End of inner exception stack trace ---
   at ScanApp.TWAIN.TwainScannerDevice.ScanPageTWAIN(Single dpi, String pixelType, String source, String payload, Scanner scanner)
   at ScanApp.Models.ScanOperation.GetPages(String id, Int32 dpi, String pixelType, String source, String payload)

Sample of my code:

```
using (Twain32 _twain32 = new Twain32())
            {
                _twain32.IsTwain2Enable = false;
                _twain32.ShowUI = false;
                _twain32.OpenDSM();

                try
                {
                    _twain32.SourceIndex = scanner.TwainIndex;
                    _twain32.OpenDataSource();
                }
                catch (TwainException ex)
                {
                    ...
                }
                _twain32.AcquireError += (sender, e) =>
                {
                    if (e.Exception.Data.Contains("FriendlyErrorMessage"))
                    {
                        TwainAcquireError = e.Exception;
                    }
                    else
                    {
                        string friendlyMessage = TwainExceptionTranslator.GetMessageFromException(e.Exception);
                        e.Exception.Data.Add("FriendlyErrorMessage", friendlyMessage);
                        TwainAcquireError = e.Exception;
                    }
                };

                _twain32.EndXfer += (sender, e) =>
                {
                    TimeSpan timeTaken = DateTime.Now - startTime;

                    string scanName = PageFileNameGenerator.CreateNewPageFileName(scanningcDir);

                    using (Image img = e.Image)
                    {
                        img.Save(scanName, ImageFormat.Jpeg);
                    }

                    // Notify about new page
                    string pageId = Path.GetFileNameWithoutExtension(scanName);
                    Events.OnPageEvent(new Events.PageInfo
                    {
                        PageId = pageId,
                        Dpi = (int)dpi,
                        PixelType = pixelType,
                        Source = source,
                        Payload = payload,
                        Scanner = scanner,
                        TimeTaken = timeTaken.ToString()
                    });

                    startTime = DateTime.Now;               // Adjust starting time for next page
                };

                this.SetDeviceSource(_twain32, source, scanner, payload);

                this.SetDeviceProperties(_twain32, dpi, pixelType, scanner, payload);
                
                _twain32.Capabilities.Indicators.Set(false);

                // Scanning starts here
                try
                {
                    startTime = DateTime.Now;
                    _twain32.Acquire();
                }
                finally
                {
                    _twain32.CloseDataSource();

                    if (TwainAcquireError != null)
                    {
                        throw TwainAcquireError;
                    }
                }
            }
        }
```

Does it seem okay to you, if I just do:
```
using (Image img = e.Image)
    {
        img.Save(scanName, ImageFormat.Jpeg);
    }
```

or am I missing something?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-22:
Hello, @MrBlueHere 
Yes of course. See #26

MrBlueHere, 2018-10-22:
Thanks, I saw the issse, but it didn't help me solve this. My app starts at 25 MB, when i launch it, but after first scan (scanning with 600dpi and flatbed), it goes to 115 MB and stays there. Weird thhing is that I don't have this problem with WIA.

MrBlueHere, 2018-10-22:
So I also tried this with OutprocSample4 and it's the same. It launeches and consumes about 15-20 MB of RAM, rises to about 600 MB when I scan at 500 dpi from flatbed and when I save the image, it falls to 100 MB. But why doesn't it fall all the way to 15-20 MB? I don't get it. When I scan with WIA, it falls back to the original size immediately.

It may not seem like an issue, but when I'm doing high speed scanning with fujitsu (around 80 pages per minute), It fails with OutOfMemory exception. when scanning at higher resolution.

Does TWAIN cause this?

MrBlueHere, 2018-10-22:
So i turned to File xFer transfer mode and there are no memory issues, right after the image is processed, memory usage falls right back to where it started. So I quess it is something that is related to Native transfer mode.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-22:
You can use a [IStreamProvider](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2411.html) if you wanna use a Native Mode Transfer. But when acquiring a large number of images or large images better use a Disk File Mode Transfer or Buffered Memory Mode Transfer. They are for this and were designed.

MrBlueHere, 2018-10-22:
Okay, I see, I should have [RTFM](https://en.wikipedia.org/wiki/RTFM). Thanks for help and have a nice day.

SaddamBInSyed, 2018-10-25:
@MrBlueHere, I am facing same  OOM issue when scanning large files, but I am disposing of the images as I instructed above and ser Xfer mode as File Xfer but STILL OOM is throwing, 

Please advise.

SaddamBInSyed, 2018-10-25:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A ,

[image]

I am using the activex websamples code, in that When I try to add the saraff.IOC dll then when launching the HTML page I am getting the above error. 

HOw do I resolve this error?
Please help.

SaddamBInSyed, 2018-10-25:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A ,
Hi saraff, 

the above issue was solved, and now when  I using the File Transfer mode , then a default file is creating in the exe location. If I comment that part then 

for(TwRC _rc=this._dsmEntry.DsRaw(this._AppId,this._srcds,TwDG.Image,TwDAT.ImageFileXfer,TwMSG.Get,IntPtr.Zero); _rc!=TwRC.XferDone; ) {
                        throw new TwainException(this._GetTwainStatus(),_rc);
                    }

the above call is failing.  

How to get rid of this,
thanks.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-10-28:
Hello, @SaddamBInSyed 
Current issue is not related to the activex websamples code. Please create separate issue.
You must add a RequiredFileAttribute to a Saraff.Twain.WebSample (in a AssemblyInfo.cs). Then rebuild a Saraff.Twain.WebSample and copy all required files to a WebSite. See a [Saraff.AxHost.NET](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/saraffaxhost/)

closed 2018-11-03 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A