saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #34: Saraff Twain SYstem.outOfMemory when aqcuire 50 pages

LuisJuarez, 2018-12-04:
HI, im using Saraff twain, it works well with 20, 30 or 40 pages, but when i put 27 sheets (54 pages) an System.OutOfMemory exception is thrown, when i debbug the instance of _twain32 have 54 pages on the buffer, how can i clean the bffer.

Im using the EndXfer evento to save the Images on the disk

`  
private void _twain32_EndXfer(object sender, Twain32.EndXferEventArgs e)
{
     using (var _image = e.Image)
     {
          _image.Save(pathImage + "\\" + imC.ToString() + ".jpeg", System.Drawing.Imaging.ImageFormat.Jpeg);
     }
     imC++;
     //e.Image.Dispose();             
}`

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-12-05:
Hello, @LuisJuarez 
You can use a [IStreamProvider](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2411.html). See https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/29#issuecomment-431940887

LuisJuarez, 2018-12-05:
Hi again, im using IStreamProvider now but the problem is the same.
[image]
twian32 instance still storing all the images that it scan, and the exception still showing.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-12-05:
If you disposing acquired image, memory will be release (even if store reference to image).
But I see that you using a MemoryStream for caching a image data. If you wanna acquiring a large number of images you must caching data of acquired image outside the RAM. See a [IStreamProvider
](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2411.html)

LuisJuarez, 2018-12-06:
thank you, now i can acquire large number of images :)

closed 2018-12-06 by LuisJuarez