saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #35: Acquired images are imcomplete

LuisJuarez, 2018-12-06:
Hi im using Memory type on XferMech Capabilitie. i can acquire a large number of images, after acquire image i store it in Disk when the acquire is completed i put the images on a PDF, but some images are damaged or incomplete.

[image]
not matter the number of pages that i acquire some times is the first page, the second, all thepages, it is completely ramdom.
Im using the "MemXferEvent" to catch the images, i put the compression capabilitie to jpeg, and the code of the MemXferEvent is here:
`try
 {
      System.Drawing.Image im = System.Drawing.Image.FromStream(new MemoryStream(e.ImageMemXfer.ImageData),true,true);
     
      im.Save(pathImage + "\\" + imC.ToString() + ".jpeg", ImageFormat.Jpeg);
      imC++;
  }`

LuisJuarez, 2018-12-06:
when i use file transfer mode the acquire method does not work, the scanner doe not recieve instruction to start scan. 
Im using Epson es-400, and kodak i2600
is it possible that it does not work with these scanners?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-12-06:
May be your scanner don't support a filexfer. You can invoke a `Twain32.Capabilities.XferMech.Get()` to get list of supported transfer modes. Also, see [Disk File Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2420.html)

LuisJuarez, 2018-12-07:
> 
> 
> Hello, @LuisJuarez
> Try directly save a ImageDate to file.
> 
> ```cs
> using(var _stream = File.Create(Path.Combine(pathImage, Path.ChangeExtension(imC, ".jpeg")))) {
>     using(var _writer = new BinaryWriter(_stream)) {
>         _writer.Write(e.ImageMemXfer.ImageData);
>     }
> }
> ```
> 
> Also, you can try a File Transfer Mode.

I tried save image directly to file but is the same problem, i realized that after the incomplete image another image is saved but i cant open this image.

[image]
is this problem caused by the memory transfer mode?
i can't try File transfer mode.

LuisJuarez, 2018-12-07:
The page 8 and 10 is the same sheet, it is like the page 8 were separated.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2018-12-08:
Hello, @LuisJuarez 
In first, see the Buffered Memory Mode Transfer (page 4-20) in the [TWAIN 2.4 Specification](https://www.twain.org/specification/) and the Transfer of Compressed Data (4-32 TWAIN 2.4 Specification).
In second, for the Buffered Memory Mode Transfer a events will be raise in follow sequence:

Twain32.SetupMemXferEvent // <<< for first page
Twain32.MemXferEvent
// …
Twain32.MemXferEvent
Twain32.XferDone

Twain32.SetupMemXferEvent // <<< for second page
Twain32.MemXferEvent
// …
Twain32.MemXferEvent
Twain32.XferDone
// ...

Iin this way, you must append several a ImageData parts in same file.

closed 2018-12-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A