saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #37: Saraff.Twain.Sample2 freezes after second scan

Nick-Burger, 2019-01-15:
I have a HP OfficeJet Pro 6970 Printer/Scanner and setup the first scan to 100dpi/Grey/Native. Using the ADF with two pages both pages where scanned perfectly as expected.
As I modified one single line the duplex property is set to false and only one side of the pages is scanned.

_twain32.Capabilities.DuplexEnabled.Set(false);

Then I place another two pages in the ADF and chage to 100dpi/RGB/Native. This two pages where also scanned as expected. But the AcquireCompleted event is never fired.
If I try to change either DPI or any other capability, the sample program freezes and the call set the new value never returns.

ThanX for your help!
Nick

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-01-15:
Hello, @Nick-Burger 
Try check expression `e.Exception.ReturnCode==TwRC.Cancel` in a _twain32_AcquireError method.

Nick-Burger, 2019-01-15:
Thank you for your quick response.
In _twain32_AcquireError e.Exception.ReturnCode never gets the value TwRC.Cancel.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-01-16:
Hello, @Nick-Burger 
Did was a _twain32_AcquireError method invoked? If yes, that what values has the properties `e.Exception.ReturnCode` and `e.Exception.ConditionCode`?

Nick-Burger, 2019-01-17:
AcquireError event is not fired.
I can get around the Problem by keeping the datasource closed. When I open the datasource right before _twain32.Acquire and close it on AcquireCompleted I can repeat these steps how ofte I want.
But keeping the datasource open like in the sample2 project will only work for the first call to _twain32.acquire. After the second call it will freeze after the last scanned page.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-01-18:
Hello, @Nick-Burger 
Highly likely, that this issue related with a datasource of your scanner (it can be not fully compliant with a twain specification). You can try send question to a hp customer service.

closed 2019-01-22 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A