saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #38: Accessing SourcesCount in TwainExternalProcess.Execute : invalid operation

SilverSnake83, 2019-01-22:
Hi

I have an issue with the saraff lib.
On the last version (**Saraff.Twain 1.0.27.606, Saraff.Twain.Aux 1.0.3.632**) When I trie to launch a **TwainExternalProcess.Execute** I get an error when accessing to **SourcesCount**.

> Operation is not valid due to the current state of the object 

```
Source _result =null;
FujitsuLogger.log(System.IO.Path.GetDirectoryName(this.GetType().Assembly.Location));
TwainExternalProcess.Execute(
	System.IO.Path.Combine(System.IO.Path.GetDirectoryName(this.GetType().Assembly.Location), Source.x86Aux),
	twain => {
		try
		{
			for (var i = 0; i < twain.SourcesCount; i++)
			{
				if (twain.GetSourceProductName(i).ToUpper().Contains(nameScanner))
				{
					_result = new Source
					{
						Id = i,
						Name = twain.GetSourceProductName(i),
						IsX64Platform = false,
						IsTwain2 = twain.IsTwain2Supported,
						IsDefault = twain.SourceIndex == i
					};
					break;
				}
			}
		}
		catch(Exception e)
		{
			FujitsuLogger.log("_GetSources : " + e.Message);
		}
	});

return _result;
```

I tried the same code with version **Saraff.Twain.Aux 1.0.2.534** and **Saraff.Twain  1.0.22.544** and now it's working fine.
The problem is once I try to do this:

```
try
{
	twain.SetCap(TwCap.XResolution, 50);
}
catch (Exception e)
{
	FujitsuLogger.log("XResolution : " + e.Message);
}
```

It gives an error 

> Data pameter out of range

And the resolution is in the list of **twain.Capabilities.XResolution.Get()**

I did manage to get the scan working for a breif amount of time, but now I always get that error.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-01-22:
Hello, @SilverSnake83 
For first part of issue, check that you has updated to latest version (Saraff.Twain 1.0.27.606, Saraff.Twain.Aux 1.0.3.632) packages in all projects (a your scan app, a Saraff.Twain.Aux_MSIL.exe and a Saraff.Twain.Aux_x86.exe). Also, you must copy a Saraff.Twain.Aux_MSIL.exe and a Saraff.Twain.Aux_x86.exe (with confing file) to a your scan app location.

For second part of issue, try change type of value for a capability value: 
`twain.SetCap(TwCap.XResolution, 50f)`. 
see [General (not strongly typed) accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2120.html). Also, you can write follow: 
`twain.Capabilities.XResolution.Set(100f)`, 
see [Strongly typed accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2110.html)

SilverSnake83, 2019-01-23:
Apparently I resolved the first part of the issue by double checking every files and config files. My bad.
The second problem was due to a missing directory in the **SetupFileXferEvent** event.
I was puting an inexistant folder in the path of **e.FileName**.
The error messages could have been more comprehensive but you helped me anyway, so thank you for your amazing work and help !

closed 2019-01-23 by SilverSnake83