saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #39: Multiscan to Page with Saraff.Twain.HtmlSamples

rahimzade, 2019-02-26:
hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 
i read [this](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25) Issue and follow all in your sample, but can not get result for multipage scan in one time. i use kodak i2620 scaner
can you help me?
thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-02-26:
Hello, @rahimzade 
Specify the question. What exactly does not work?

rahimzade, 2019-02-26:
when i change sample, i get this error on load capabilities. 
[image]
i can scan multipage in windows by this scaner

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-02-27:
Hello, @rahimzade 
Try check your scanner using a Saraff.Twain.Sample1 and a Saraff.Twain.Sample2 of a [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

rahimzade, 2019-02-27:
> 
> 
> Hello, @rahimzade
> Try check your scanner using a Saraff.Twain.Sample1 and a Saraff.Twain.Sample2 of a [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

hello @saraff-9EB1047A4BEB4cef8506B29BA325BD5A
 i checked my scanner in [this link](http://asprise.com/scan/scannerjs/demo/scanner.js-demo.php?license=Enterprise_10Yrs) and this work fine. but when check in  [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html) samples, after scan first page scanner are worked continuously and not scan second paper!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-02-27:
An Saraff.Twain.NET CS Samples is scan only one page per click (session). However, you need to check your scanner using a Saraff.Twain.Sample1 and a Saraff.Twain.Sample2, because, there is a suspicion, that your scanner don't support some capabilities (in particular, ICAP_PIXELTYPE (The type of pixel data: B/W, gray, color, etc.), ICAP_XRESOLUTION (Current/Available optical resolutions for x-axis) and  ICAP_YRESOLUTION (Current/Available optical resolutions for y-axis))

closed 2019-03-05 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

mehmetfatihustdag, 2019-03-08:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A , 

I am trying to call api http://localhost:8081/twain-web/GetSources. After install windows service from the Saraff.Twain.Service project.  

[image] 

Windows service as running as follows : 
[image]

However api returns 400 (Bad Request Error ) . Once i installed windows service from installer package (Saraff.Twain.Service_x86) .   Api works. 
[image]

Can you help me ? How can i resolve this problem?
Thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-03-08:
Hello, @mehmetfatihustdag 
You can try follow: add try-catch to a Saraff.Twain.Aux_MSIL (and Saraff.Twain.Aux_x86). [See this topic](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/14#issuecomment-384591012)

mehmetfatihustdag, 2019-03-09:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A  I tried but nothing changed. After calling http://localhost:8081/twain-web/GetSource , i checked the log file as follows 

Handling an exception.  Exception details: System.ServiceModel.FaultException: Win32Exception: CreateProcessAsUser failed.
   at Saraff.Twain.Aux.ImpersonationUtils._LaunchProcessAsUser(String fileName, IntPtr token, IntPtr envBlock, IntPtr&amp; hStdIn, IntPtr&amp; hStdOut, IntPtr&amp; hStdError)
   at Saraff.Twain.Aux.ImpersonationUtils.RunAsCurrentUser(String fileName)
   at Saraff.Twain.Aux.TwainExternalProcess.AuxProcess.CreateProcess(String fileName)
   at Saraff.Twain.Aux.TwainExternalProcess.Execute(String fileName, Action`2 execCallback)
   at Saraff.Twain.Service.TwainService.GetSources() in C:\Users\f.ustdag\Desktop\Experimental\sarraff\Saraff.Twain.HtmlSamples\Saraff.Twain.HtmlSamples\Saraff.Twain.Service\TwainService.svc.cs:line 72
   at Saraff.Twain.Service.TwainService.GetSources() in C:\Users\f.ustdag\Desktop\Experimental\sarraff\Saraff.Twain.HtmlSamples\Saraff.Twain.HtmlSamples\Saraff.Twain.Service\TwainService.svc.cs:line 97

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-03-10:
Hello, @mehmetfatihustdag 
Has you updated a Saraff.Twain.NET package in some projects of a Saraff.Twain.HtmlSamples solution?

mehmetfatihustdag, 2019-03-20:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A , İ  updated, and problem has gone. Thank you. 
I have another question. When there is no paper on the scanner, i want to show error like (No page on the scanner , add page the scanner ) instead of Error Occured  . Is it possible to do this with html sample project ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-03-21:
Hello, @mehmetfatihustdag 
Yes, you can. You need to overwrite a catch (or add a TwainException catch) in a `public Stream Acquire(Source source,IEnumerable<CapEnum> caps)` method of a Saraff.Twain.Service.TwainService class.

mehmetfatihustdag, 2019-04-17:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A , i tried to add catch TwainException catch. Once i tried with scenerio (Paper Jam) . Pop up appeared like above 
[image]. I think because of scanner provide that pop ups. However, i would like to see error cases (Paper Jam, Double Paper Feed ext ) returning from rest api (acquire). However, it returns success in the error cases . When i debugged catch(TwainException e ) didnt catch these exceptions.  I am using **Canon dr-c240 Scanner** 

Here is my code : 

` public Stream Acquire(Source source, IEnumerable<CapEnum> caps)
       {
           try
              {
                var _formatter = new BinaryFormatter();

                var _result = new MemoryStream();
                var _images = new List<System.Drawing.Image>();
                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain =>
                    {
                        twain.EndXfer += (sender, e) => _images.Add(e.Image);
                        twain.SourceIndex = source.Id;
                        twain.OpenDataSource();

                        foreach (var _cap in caps)
                        {
                            
                            using (var _stream = new MemoryStream(_cap.Current.RawValue))
                            {
                                twain.SetCap(_cap.Cap, _formatter.Deserialize(_stream));
                            }
                        }
                       
                            twain.Acquire();
                    });
                var _params = new EncoderParameters(2);
                _params.Param[0] = new EncoderParameter(System.Drawing.Imaging.Encoder.Compression, (long)EncoderValue.CompressionNone);
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.MultiFrame);
				
               _images[0].Save(
                   _result, ImageCodecInfo.GetImageEncoders().First(x => x.MimeType == "image/tiff"), _params);
              
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.FrameDimensionPage);
                 for (int i = 0; i < _images.Count; i++)
                 {
                      _images[0].SaveAdd(_images[i], _params);
                 }
                foreach (var _image in _images)
                {
                    _image.Dispose();
                }

               _result.Seek(0, SeekOrigin.Begin);
                return _result;
            }catch(TwainException twainException)
            {
                throw twainException;

            }
            catch (Exception ex)
            {
         
                throw new FaultException(ex.Message);
            }
			
	}`

Could you give me an idea ? 
Thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-04-17:
Hello, @mehmetfatihustdag 
In first, add a Twain32.AcquireError handler to your code:
```c#
                // …

                TwainException _AcquireException = null;

                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain => {
                        twain.EndXfer+=(sender,e) => _images.Add(e.Image);

                        twain.AcquireError += (sender, e) => _AcquireException = e.Exception;

                        // ...
                  });
                if(_AcquireException != null) {
                    if(_AcquireException.ConditionCode == TwCC.PaperJam) {
                        throw new InvalidOperationException("PaperJam", _AcquireException);
                    }
                    throw new InvalidOperationException(_AcquireException.Message, _AcquireException);
                }
                
                // ...
```
In second, you must throw only FaultException:
```c#
            try {

                // …

            } catch(TwainException ex) {
                if(ex.ConditionCode == TwCC.PaperJam) {
                    throw new FaultException("PaperJam");
                }
                var _message = string.Empty;
                for(Exception _ex = ex; _ex != null; _ex = _ex.InnerException) {
                    _message += string.Format("{0}: {1}{3}{2}{3}{3}", _ex.GetType().Name, _ex.Message, _ex.StackTrace, Environment.NewLine);
                }
                throw new FaultException(_message);
            } catch(Exception ex) {
                var _message=string.Empty;
                for(var _ex=ex; _ex!=null; _ex=_ex.InnerException) {
                    _message+=string.Format("{0}: {1}{3}{2}{3}{3}",_ex.GetType().Name,_ex.Message,_ex.StackTrace,Environment.NewLine);
                }
                throw new FaultException(_message);
            }

```

Oscar1and, 2023-03-24:
Estoy utilizando el método http://localhost:8081/twain-web/Acquire le puedo enviar los valores de :
?platform=X86_32&version=V1&sourceId=1&caps=%7B%22IPixelType%22:%22%5B0,1,0,0,0,255,255,255,255,1,0,0,0,0,0,0,0,4,1,0,0,0,13,83,121,115,116,101,109,46,85,73,110,116,49,54,1,0,0,0,7,109,95,118,97,108,117,101,0,14,1,0,11%5D%22,%22XResolution%22:%22%5B0,1,0,0,0,255,255,255,255,1,0,0,0,0,0,0,0,4,1,0,0,0,13,83,121,115,116,101,109,46,83,105,110,103,108,101,1,0,0,0,7,109,95,118,97,108,117,101,0,11,0,0,150,67,11%5D%22,%22YResolution%22:%22%5B0,1,0,0,0,255,255,255,255,1,0,0,0,0,0,0,0,4,1,0,0,0,13,83,121,115,116,101,109,46,83,105,110,103,108,101,1,0,0,0,7,109,95,118,97,108,117,101,0,11,0,0,150,67,11%5D%22,
pero no se como adicionarle el valor para duplex, intente con FeederEnabled, pero genera un error como se puede concatenar ese valor