saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #40: Save multiple scans in single TIFF

QueenPirate, 2019-02-28:
Good Day, Saraff,

We used the Saraff.Twain.HtmlSamples and were able to scan multiple pages in our scanner. However, it is only saving the last scanned document when uploaded. 

Below are the changes we did.

This is from jtwain.js:

$("#notify").text("Ready.");
 $("#image").prop("src", "data:image/tiff;base64," + _image);  //from jpeg to tiff
$("#image").data({ base64: _image });

This is from TwainService.svc.cs under Acquire:

_image.Save(_result, ImageFormat.Tiff); //From jpg to tiff

Are there any other changes to do? Thank you.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-02-28:
Hello, @QueenPirate 
Try see a [Multiscan to Page issue](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/25)

QueenPirate, 2019-03-01:
Yes, I actually followed that. What you meant when directly saving the multi-page tiff to a server is saving it via TwainService and not via upload from jtwain.js with UploadHandler.ashx?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-03-01:
Hello, @QueenPirate 
In this cause a TwainService directly save the multi-page tiff to a server (when scanning complete) upload via a UploadHandler.ashx, but it not good solution.

QueenPirate, 2019-03-05:
Hi, Saraff,

I managed to create a class for merging multi-page tiff and saving in directory of TwainService. Then return its base64 to server-side web page. Thank you very much and more power! :)

closed 2019-03-05 by QueenPirate