saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #41: Out Of Memory exception when scanning several pages

KrzysztofKielce, 2019-03-15:
Hi Saraff Team,

I got an error on one of my customers' site where the scanning of multiple images throws an Out Of Memory exception. I managed to reproduce it on my development machine and the issue seems to be related to the RAM memory being fragmented and errors when allocating 60 MB byte array. How it could be fixed/workarounded?

SOME MORE DETAILS:

The error is thrown while assigning data to a memory stream. What is odd, the whole application process takes less than 400 MB RAM:
[image]

I have decompiled Saraff.Twain.dll, added some debug info there, compiled it again and replaced  the dll in the application:
[image]

….and it seems that the faulty memory stream is only 33 MBs long. Below is the moment when the stream is going to be allocated to another place in memory and doubled in size:
[image]

What is odd, the first image stream size succeed to double to 60 MB:

DEBUG BufferSize=262144, size=38340904, num=262144, i=33554432,   Stream: Position=33554448, Capacity = **33556480**, Length = 33554448
DEBUG BufferSize=262144, size=38340904, num=262144, i=33816576,   Stream: Position=33816592, Capacity =**67112960**, Length = 33816592
DEBUG BufferSize=262144, size=38340904, num=262144, i=34078720,   Stream: Position=34078736, Capacity = 67112960, Length = 34078736
DEBUG BufferSize=262144, size=38340904, num=262144, i=34340864,   Stream: Position=34340880, Capacity = 67112960, Length = 34340880
(...)
DEBUG BufferSize=262144, size=38340904, num=262144, i=38010880,   Stream: Position=38010896, Capacity = 67112960, Length = 38010896
DEBUG BufferSize=262144, size=38340904, num=67880, i=38273024,   Stream: Position=38273040, Capacity = z, Length = 38273040

It does not occur when scanning a single page. Only multiple pages cause it.

The issue might lay in Saraff.Twain._ImageHandler.PtrToStreamCore() method,
 _writer.Write(_buffer,0,_len);  line of code.

Your help will be strongly appreciated.

Cheers,
Krzys

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-03-15:
Hello, @KrzysztofKielce 
In first, you can override processing a acquired image data from unmanaged memory to a stream. To do this, you need to implement the IImageHandler interface and place it to the IoC container. This feature is available starting with a Saraff.Twain.NET 1.0.26.605 (see [Using a IImageHandler for processing acquired a image data from unmanaged memory to a stream](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2412.html)).
In second, you can use a [Buffered Memory Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2430.html) or [Disk File Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2420.html).

KrzysztofKielce, 2019-03-19:
Thanks a tone!
Your first suggestion (IoC) worked for me best.

closed 2019-03-26 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A