saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #43: Scan Pages By one by

hakandeneme1234, 2019-04-30:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A , 

I am trying to use your library in order to scan multiple pages . When i try to call the method, i can get pages at the end with a tiff file format. However, i would like to get pages one by one while scanning process continue. How i can achieve ? Is there a way by using saraff twain?

Thanks

` public Stream Acquire(Source source, IEnumerable<CapEnum> caps)
        {
            var _result = new MemoryStream();

            try
            {
                var _formatter = new BinaryFormatter();

                var _images = new List<System.Drawing.Image>();
                TwainExternalProcess.Execute(
                    TwainService._GetExecFileName(source),
                    twain => {
                        twain.ShowUI = false;
                        twain.EndXfer += (sender, e) => _images.Add(e.Image);
                        twain.SourceIndex = source.Id;
                        twain.OpenDataSource();

                        foreach (var _cap in caps)
                        {
                            using (var _stream = new MemoryStream(_cap.Current.RawValue))
                            {
                                twain.SetCap(_cap.Cap, _formatter.Deserialize(_stream));
                            }
                        }

                        twain.Capabilities.XferCount.Set(-1);
                        twain.Capabilities.Indicators.Set(false);
                        twain.Capabilities.FeederEnabled.Set(true);

                     
                        twain.Acquire();
                    
                    });

              
                var _params = new EncoderParameters(2);
                _params.Param[0] = new EncoderParameter(System.Drawing.Imaging.Encoder.Compression, (long)EncoderValue.CompressionNone);
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.MultiFrame);

                _images[0].Save(
                    _result, ImageCodecInfo.GetImageEncoders().First(x => x.MimeType == "image/tiff"), _params);
                                                                     
                _params.Param[1] = new EncoderParameter(System.Drawing.Imaging.Encoder.SaveFlag, (long)EncoderValue.FrameDimensionPage);

                for (int i = 1; i < _images.Count; i++)
                {
                    _images[0].SaveAdd(_images[i], _params);
                }
                foreach (var _image in _images)
                {
                    _image.Dispose();
                }

            _result.Seek(0, SeekOrigin.Begin);
            return _result;

        }
           
            catch (Exception ex)
            {
              
                var _message = string.Empty;
                for (var _ex = ex; _ex != null; _ex = _ex.InnerException)
                {
                    _message += string.Format("{0}: {1}{3}{2}{3}{3}", _ex.GetType().Name, _ex.Message, _ex.StackTrace, Environment.NewLine);
                }
                throw new FaultException(_message);
            }
          
        }`

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-04-30:
Hello, @hakandeneme1234 
See a #25 issue (Multiscan to Page) and a #39 issue (Multiscan to Page with Saraff.Twain.HtmlSamples)

closed 2019-05-06 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

rmarques14, 2019-09-30:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A ,

I am using your library in a WinForm application and i'm struggling to implement a multi scan method (page by page) using the horizontal scanner not the feeder. I've read issues #39 and #25 but still, I can't get it. 

When I call the __twain.Acquire()_ method inside a FOR loop, it only scans the first time. The _XferCount_ cap is set to -1 and the i've also set to false the _DisableAfterAcquire_ property. 

```
PAGE_NUM = Int32.Parse(metroTextBoxPageNum.Text);
for (var i = 1; i <= PAGE_NUM; i++)
{
     _twain.Acquire();
}
```

In one of your samples [https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/threading.html](url), it seems to be a sample where you define the number of documents to be scanned but, when I download the zip with the source code, none of the projects corresponds to the **Figure 2 - Saraff.Twain.MultithreadingSample1** 

Is there any way I can do to achive a page by page scan ? Can you help me with this ? 

Thank you ;)