saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #48: FileDisk Mode: How do I stop the creation of multiple empty tmp files?

bkraul, 2019-09-05:
I am using the Saraff.Twain.Extensions. I am using TwSX.File as my transfer mode, which works great. However, even when I specify the file name on `setupCallback`, I notice tmp files such as are getting created in the temp folder. I do not have any code that would be generating that temp file name.

[image]

Is there any way to disable this behavior?

bkraul, 2019-09-05:
It seems this is found on line [1167](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/blob/1d0a68031993f53762a1d99755a5bfff5235d1f5/Twain32.cs#L1167) of `Twain32.cs`. I do not see a way to suppress, without changing the source, though.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-09-06:
HelloThe Path.GetTempFileName() don't create any file.

bkraul, 2019-09-06:
This could work:
```cs
string tempFileName = Path.GetTempFileName();
File.Delete(tempFileName);

TwSetupFileXfer _fileXfer=new TwSetupFileXfer {
    Format=((this.Capabilities.ImageFileFormat.IsSupported()&TwQC.GetCurrent)!=0)?this.Capabilities.ImageFileFormat.GetCurrent():TwFF.Bmp,
    FileName=tempFileName
};
```

bkraul, 2019-09-06:
I'm afraid it does, man, it's on the official documentation.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-09-06:
Hello, dear @bkraul
In the first lines of my letter I report the following:
> This could work:
> string tempFileName = Path.GetTempFileName();
> File.Delete(tempFileName);
> 
> TwSetupFileXfer _fileXfer=new TwSetupFileXfer {
>     Format=((this.Capabilities.ImageFileFormat.IsSupported()&TwQC.GetCurrent)!=0)?this.Capabilities.ImageFileFormat.GetCurrent():TwFF.Bmp,
>     FileName=tempFileName
> };

This is interesting, but you could write better.

See the  https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/commit/c734c24a8037f92ee6b96f103548f490bf9cb340 commit, please.
Try check this changes. And I will publish new version of the nuget package.

Best Regards

bkraul, 2019-09-06:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A, it seems that would work as well, as it would only create the temp file if no filename is passed in the arguments, which would be unlikely.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-09-08:
Hello, @bkraul
See the [Saraff.Twain.NET 1.0.28.662](https://www.nuget.org/packages/Saraff.Twain/1.0.28.662) NuGet package.

closed 2019-09-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A