saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #50: Multi Scan (Page by Page) C# App

rmarques14, 2019-09-30:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A ,

I am using your library in a WinForm application and i'm struggling to implement a multi scan method (page by page) using the horizontal scanner not the feeder. I've read issues #39 and #25 but still, I can't get it. 

When I call the __twain.Acquire()_ method inside a FOR loop, it only scans the first time. The _XferCount_ cap is set to -1 and the i've also set to false the _DisableAfterAcquire_ property. 

```
PAGE_NUM = Int32.Parse(metroTextBoxPageNum.Text);
for (var i = 1; i <= PAGE_NUM; i++)
{
     _twain.Acquire();
}
```

In one of your samples [https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/threading.html](url), it seems to be a sample where you define the number of documents to be scanned but, when I download the zip with the source code, none of the projects corresponds to the **Figure 2 - Saraff.Twain.MultithreadingSample1** 

Is there any way I can do to achive a page by page scan ? Can you help me with this ? 

Thank you ;)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-09-30:
Hello, @rmarques14 
Unacceptable invoke the Twain32.Acquire() method in loop, because it is async method (for winform apps).
You can try follow ways: set the XferCount = PAGE_NUM; Or invoke the Twain32.Acquire() method from the Twain32.EndXfer event handler.

rmarques14, 2019-09-30:
I've tried both solutions and nothing happens, only scans the first time I invoke the Twain32.Acquire() method...

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-09-30:
I checked - it works
```c#
private void _twain_AcquireCompleted(object sender,EventArgs e) {
    try {
        if(expression) {
            this._twain.Acquire();
        }
    } catch(Exception ex) {
        // ...
    }
}

private void _twain_EndXfer(object sender, Twain32.EndXferEventArgs e) {
    try {
        if(this.pictureBox1.Image != null) {
            this.pictureBox1.Image.Dispose();
        }
        this.pictureBox1.Image = e.Image;
    } catch(Exception ex) {
        // ...
    }
}
```

rmarques14, 2019-10-02:
Hi @saraff-9EB1047A4BEB4cef8506B29BA325BD5A,

You're right, it works !! I was calling the _this._twain.Acquire()_ method without using a Try...Catch block, I'm so dumb hahahaha

Thanks for the help !!!

closed 2019-10-02 by rmarques14