saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #51: too many images on memxfer

LuisJuarez, 2019-10-07:
Hi, im using XferMech capabilitie as memory, my problem is when th resolution of tue scanner is chenged, with 70 dpi on x and y resolution works fine but the images are so blurred, if i increase the dpi the result of images also increases. the number of images increases depending the number of dpi.

Can you help me to fix it?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-08:
Hello, @LuisJuarez 
If I understand correctly, then the problem is related to the physical restriction of your scanner (70 dpi).
In this case, the Saraff.Twain.NET cannot influence the situation.

LuisJuarez, 2019-10-08:
But, in the epson software i can scan up to 1200 dpi, and only 1 image is generated.
I dont think it is because the scaner restrictions.

I attach some images to explain this better.

[image]
[image]
[image]

In the first two images belong to he first scanned page, the next images belong to next pages (the last images are OK),

LuisJuarez, 2019-10-08:
i just scanned 3 pages on one side

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-08:
Hello, @LuisJuarez 
Try using the Saraff.Twain.Sample2 from the [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html) (select required xfexmech and dpi)

closed 2019-10-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A