saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #52: "Twain32.ImageInfo" doesn't contain description for "SamplesPerPixel"

KureninovVS, 2019-10-25:
Hi, sorry if it is lame mistake or something, tried to use code on "Buffered Memory Mode Transfer".

[image]

VS tells me there is no such member (Error CS1061). I am only using Saraff.Twain 1.0.28.662. Do i need more packages?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-25:
Hello, @KureninovVS 
The SamplesPerPixel has removed at the ccf8be07179e5a4414a3190b5f83dd7dfccfe27a revision (see Twain32.cs file). Use the BitsPerSample.Length instead.
I has fixed the wiki page.

KureninovVS, 2019-10-28:
Many thanks! I am newbie, so I was felling myself real dummy at this point, you saved me )

closed 2019-10-28 by KureninovVS