saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #53: Problem with "Close" and "Cancel" option

KureninovVS, 2019-10-28:
Hi, it's me again 0/. I have been using  #`Twain32.Acquire()` with  `Twain32.IsTwain2Enable = true;` and found something strange. On "Close" and "Cancel" buttons form stucks and form time to time fires System.AccessViolationException with source "System.Windows.Forms".
Found StackTrace.

> in System.Windows.Forms.UnsafeNativeMethods.DispatchMessageA(MSG& msg) in System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData) in System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context) in System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context) in Saraff.Twain.Twain32.Acquire() in ScanService.Logic.ScannerLogic.FetchPDF() in C:\Users\user\Documents\scanservice\ScanService\Logic\ScannerLogic.cs:line 150 in ScanService.Logic.ScannerLogic.GeneratePdf(Nullable`1 pageLimitOrNull) in C:\Users\user\Documents\scanservice\ScanService\Logic\ScannerLogic.cs:line 301 in ScanService.Controllers.ScanController.GetPdf(Nullable`1 maxPages) in C:\Users\user\Documents\scanservice\ScanService\Controllers\ScanController.cs:line 86 in lambda_method(Closure , Object , Object[] ) in System.Web.Http.Controllers.ReflectedHttpActionDescriptor.ActionExecutor.<>c__DisplayClass6_2.b__2(Object instance, Object[] methodParameters) in System.Web.Http.Controllers.ReflectedHttpActionDescriptor.ActionExecutor.Execute(Object instance, Object[] arguments) in System.Web.Http.Controllers.ReflectedHttpActionDescriptor.ExecuteAsync(HttpControllerContext controllerContext, IDictionary`2 arguments, CancellationToken cancellationToken) in System.Web.Http.Controllers.ApiControllerActionInvoker.d__1.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.Start[TStateMachine](TStateMachine& stateMachine) in System.Web.Http.Controllers.ApiControllerActionInvoker.InvokeActionAsyncCore(HttpActionContext actionContext, CancellationToken cancellationToken) in System.Web.Http.Controllers.ApiControllerActionInvoker.InvokeActionAsync(HttpActionContext actionContext, CancellationToken cancellationToken) in System.Web.Http.Controllers.ActionFilterResult.d__5.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.Start[TStateMachine](TStateMachine& stateMachine) in System.Web.Http.Controllers.ActionFilterResult.ExecuteAsync(CancellationToken cancellationToken) in System.Web.Http.ApiController.ExecuteAsync(HttpControllerContext controllerContext, CancellationToken cancellationToken) in System.Web.Http.Dispatcher.HttpControllerDispatcher.d__15.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.Start[TStateMachine](TStateMachine& stateMachine) in System.Web.Http.Dispatcher.HttpControllerDispatcher.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Net.Http.HttpMessageInvoker.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Web.Http.Dispatcher.HttpRoutingDispatcher.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Net.Http.DelegatingHandler.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Web.Http.HttpServer.d__24.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.Start[TStateMachine](TStateMachine& stateMachine) in System.Web.Http.HttpServer.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Net.Http.HttpMessageInvoker.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in System.Web.Http.Owin.HttpMessageHandlerAdapter.d__20.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine](TStateMachine& stateMachine) in System.Web.Http.Owin.HttpMessageHandlerAdapter.InvokeCore(IOwinContext context, IOwinRequest owinRequest, IOwinResponse owinResponse) in System.Web.Http.Owin.HttpMessageHandlerAdapter.Invoke(IOwinContext context) in Microsoft.Owin.Infrastructure.OwinMiddlewareTransition.Invoke(IDictionary`2 environment) in Microsoft.Owin.Hosting.Utilities.Encapsulate.Invoke(IDictionary`2 environment) in Microsoft.Owin.Host.HttpListener.OwinHttpListener.d__30.MoveNext() in System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine](TStateMachine& stateMachine) in Microsoft.Owin.Host.HttpListener.OwinHttpListener.ProcessRequestAsync(HttpListenerContext context) in Microsoft.Owin.Host.HttpListener.OwinHttpListener.d__29.MoveNext() in System.Runtime.CompilerServices.AsyncMethodBuilderCore.MoveNextRunner.InvokeMoveNext(Object stateMachine) in System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx) in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx) in System.Runtime.CompilerServices.AsyncMethodBuilderCore.MoveNextRunner.Run() in System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c__DisplayClass4_0.b__0() in System.Runtime.CompilerServices.AsyncMethodBuilderCore.ContinuationWrapper.Invoke() in System.Runtime.CompilerServices.TaskAwaiter.<>c__DisplayClass11_0.b__0() in System.Runtime.CompilerServices.AsyncMethodBuilderCore.ContinuationWrapper.Invoke() in System.Threading.Tasks.AwaitTaskContinuation.RunOrScheduleAction(Action action, Boolean allowInlining, Task& currentTask) in System.Threading.Tasks.Task.FinishContinuations() in System.Threading.Tasks.Task.FinishStageThree() in System.Threading.Tasks.Task`1.TrySetResult(TResult result) in System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization) in System.Threading.Tasks.TaskFactory`1.<>c__DisplayClass35_0.b__0(IAsyncResult iar) in System.Net.LazyAsyncResult.Complete(IntPtr userToken) in System.Net.LazyAsyncResult.ProtectedInvokeCallback(Object result, IntPtr userToken) in System.Net.ListenerAsyncResult.IOCompleted(ListenerAsyncResult asyncResult, UInt32 errorCode, UInt32 numBytes) in System.Net.ListenerAsyncResult.WaitCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* nativeOverlapped) in System.Threading._IOCompletionCallback.IOCompletionCallback_Context(Object state) in System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx) in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx) in System.Threading._IOCompletionCallback.PerformIOCompletionCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* pOVERLAP)

It is strange because if I first use "Scan" button (without paper) and then use "Cancel" or "close window" options it close smooth with no problem... Can this problem be part of what  my app targeted to 4.6.1 .NET? Also I use it in service, so can there be something  what i should be aware of like requirement of multithreading?

KureninovVS, 2019-10-28:
oh, and also, have on idea if this is connected, but form doesn't appear on task bar.(Using windows 10)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-28:
Hello, @KureninovVS 
About the AccessViolationException, likely this related with your scanner (scanner driver).
About the service, see the [Saraff.Twain.NET Service Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/service.html).

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-28:
Also you may see the [Saraff.Twain.NET HTML Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/html.html)

KureninovVS, 2019-10-28:
If this problem part of scanner driver, does that mean it is just unsolvable issue for me? I am using CANON DR-M160II, and the is only on driver of for windows device...

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-28:
Obviously the scanner driver was not ready for multithreaded calls.
See the Saraff.Twain.NET Service Samples.

KureninovVS, 2019-10-28:
Okay, I'll try that ). I'll notify of my progress here

KureninovVS, 2019-10-28:
Well, apparently i stuck with no match exception(ServiceSample4):
[image]
Am I missing something?(I installed Saraff.Twain.Service_x64.msi)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-10-28:
Из трассировки стека видно, что используемый по умолчанию источник данных (сканер)  вернул пустой список поддерживаемых разрешений для сканирования (dpi).
```c#
var _caps = await this.TwainService.GetCapsAsync(_ds,new List<TwCap> { TwCap.XResolution,TwCap.IPixelType });
this._Fill<float>(this.resolutionsToolStripDropDownButton,_caps.First(val => val.Cap==TwCap.XResolution));
```
Соответственно метод `First` выдал исключение. 
Если источник данных не возвращает список поддерживаемых разрешений (dpi), то он не соответствует спецификации TWAIN. Попробуйте использовать другой драйвер для сканера.
Также вы можете проверить работу указанного сканера с помощью приложения Saraff.Twain.Sample2 (из [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html) ), я думаю результат будет тем же.

KureninovVS, 2019-10-28:
Saraff.Twain.Sample2 и Saraff.Twain.Sample1 не распознают устройство как сканер... Я установил другой дравер, но он ведет себя также... Хотя оба они себя катируют как ISIS / TWAIN Driver... Ладно, большое спасибо за помощь) я не думаю что буду дальше решать этот вопрос, просто отключу пользовательский интерфейс и поставлю нужную конфигурацию сканирования для источника

closed 2019-10-28 by KureninovVS