saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #54: save pdf file on server Saraff.Twain.HtmlSamples

Dirceu-Oliveira, 2019-11-07:
Hello, I'm getting the image, but I can't save the file in pdf, would it be in this function of jQuery?

        $("#upload").click(function () {
            let _image = $("#image").data()["base64"];
            if (_image === undefined) {
                alert("The image does not acquired yet.");
                return;
            }
            $("#toolstrip").children().prop("disabled", true);
            $("#notify").text("uploading...");
            $.post("uploadhandler.ashx", { action: "create", ext: "jpg" }, function (filename) {
                let _append = function (position,blockSize) {
                    $("#notify").text("uploading... " + filename + " " + Math.round(position / _image.length * 100)+"%");
                    $.post("uploadhandler.ashx", { action: "append", name: filename, data: _image.substr(position, blockSize) })
                        .done(function () {
                            if (position + blockSize < _image.length) {
                                _append(position + blockSize, blockSize);
                            } else {
                                $("#toolstrip").children().prop("disabled", false);
                                $("#notify").text("uploaded. " + filename);
                            }
                        })
                        .fail(function () {
                            $("#toolstrip").children().prop("disabled", false);
                            $("#notify").text("upload failed. " + filename);
                        });
                };
                _append(0, 8 * 1024);
            }).fail(function () {
                $("#toolstrip").children().prop("disabled", false);
                $("#notify").text("upload failed.");
            });
        });

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-11-08:
Hello, @Dirceu-Oliveira 
In JQuery save to a pdf impossible because JS haven't access to a file system. When uploaded to web site, you can save to a pdf. For this purpose use the iTextSharp.

Dirceu-Oliveira, 2019-11-09:
Where will I enter the iTextSharp code in my controller or UploadHandler.ashx?
Would you have any code as an example?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-11-09:
Hello, @Dirceu-Oliveira 

```c#
Document doc = new Document();
PdfWriter.GetInstance(doc, new System.IO.FileStream(this._fileName, System.IO.FileMode.Create));
doc.Open();
for (int i = 0; i <= Scanners.Twain.ImageCount - 1;i++ ) {
    iTextSharp.text.Image img = iTextSharp.text.Image.GetInstance(Scanners.Twain.GetImage(i), System.Drawing.Imaging.ImageFormat.Jpeg);  
    doc.SetPageSize(new iTextSharp.text.Rectangle(0, 0, img.Width, img.Height));
    doc.NewPage();
    doc.Add(img);
}
doc.Close();   
```

Also you can see follow links
https://simpledotnetsolutions.wordpress.com/2012/04/08/itextsharp-few-c-examples/
https://github.com/itext/itextsharp
https://github.com/itext/itext7-dotnet

closed 2019-11-14 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

Dirceu-Oliveira, 2019-11-14:
Hi, I was able to convert the jpg to pdf, now I need
 include in HtmlSamples buttons to delete and rotate scanned images?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-11-14:
Hello, @Dirceu-Oliveira 
You wrote: "include in HtmlSamples buttons to delete and rotate scanned images?"
You can do it yourself, because this is beyond the scope of the Saraff.Twain.NET project.