saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #55: "DG DAT MSG out of expected sequence" when Acquire() with official TWAINDSM

MakiseKurisu, 2019-11-07:
We are currently trying to convert a 32-bit program to 64-bit. However, this breaks our scanning code. The changes we made to the code are switching from 32-bit to 64-bit and Twain v1 to Twain v2 as v1 was not available under x64. If we revert those changes then the code will work again. Here is all the possible combinations that we have tried:
[image]

So this appears to be an issue with TWAINDSM. We are using v2.4.2 from https://github.com/twain/twain-dsm/tree/master/Releases/dsm_020402/windows so it should be official. Scanner is Panasonic KV-S1057C with the latest TWAIN driver. Windows was also reinstalled in last week.

We also tried Saraff.Twain.Sample1 that came with Saraff.Twain.Extension sample. 32-bit version works fine but 64-bit version also has the same issue:
[image]

Do you think if this is a driver issue? If so, what can we do here to keep the program in 64-bit?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-11-08:
Hello, @MakiseKurisu 
Likely that the TWAINDSM.dll (32 and 64 bits) incorrectly installed on your PC. See the [Installation of the Data Source Manager](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w0200.html) wiki page.

closed 2019-11-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

GuyTraveler, 2024-05-28:
I hate to resurrect this issue but I thought I would apply what I learned. If you investigate you will find that the scanner / TWAIN interaction is in the correct state and that your callback receives a ready status. The issue is that the callback expects you to take action outside of the callback procedure. You can see this in the DSM_Null method of the TWAINDSM source code.

    // Set flag to prevent Application from sending a new message 
    // before returning back from recieving this callback.
    pod.m_ptwndsmapps->DsSetAppProcessingCallback(_pAppId,(TWID_T)_pDsId->Id,TRUE);
    
The flag does not get reset until the callback finalizes. (Your code completes executing.) As a workaround I wrapped my transfer code in a Task.Run to delay execution and avoid a more complicated implementation.