saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #56: Scanner connection events

MrBlueHere, 2019-12-06:
Hi,

is it possible to detect if some scanner connected / disconnected from the device, so that the list of scanners doesn't have to be checked manually?

For example some Event that I could register an EventHandler to and update the list of scanners when it fires.

Thanks.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-06:
Hello, @MrBlueHere 
It's impossible. See [TWAIN 2.4 Specification](https://twain.org/specification/)

MrBlueHere, 2019-12-07:
Thanks for a quick reply. I've checked the specification and I found an interesting event: TWDE_DEVICEREMOVED. As I understand it, this event should fire when the scanner disconnects. Isn't it possible to bind to this event so that I get notified when the scanner disconnects?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-08:
Hello, @MrBlueHere 
For this target you can use the `Twain32.DeviceEvent`. But it available only after that you opening DS and if you DS supported it.
```C#
this._twain32.DeviceEvent += (sender, e) => {
    switch(e.Event) {
        case TwDE.DeviceRemoved:
            // <<<
            break;
    }
};
```
Also see the CAP_DEVICEEVENT (10-40 TWAIN 2.4 Specification)

closed 2019-12-13 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A