saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #57: 2 Versions of Scanner

Ponkie, 2019-12-17:
Hi Saraff,

I'm using a Alaris E1035 scanner. On GetSources, the program is getting a 32 and 64-bit version of the scanner (And WIA too, but let's ignore that).

I installed the 64-bit service but I'm not getting any result when using the X86_64 one. 
The acquire process is just pending forever and not returning anything. The 32 and 64 bit one also share the same Id, and IsDefault both are true. 
I already tried using a variable to have them different Id's but, I can't get any results for GetCaps on the other one if they have the different Ids.

I can get results on the 32-bit just fine though. 

Any insights on why this is happening? I'd like to have the 64-bit one working as well just in case.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-17:
Hello, @Ponkie 
Specify what a code examples you use. I can assume that this is the Saraff.Twain.NET Service Samples (or Saraff.Twain.NET Outproc Samples).
In this case, the 32 and 64 bits versions must have the same Id, and IsDefault both are true, because each host (32 and 64 bits) return own collection of a DS.

You wrote: "Any insights on why this is happening? I'd like to have the 64-bit one working as well just in case". Likely that problem in your driver of scanner (64 bits). Try check out it using the Saraff.Twain.Sample2 from [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

About UI language, you can configure it in the Twain32.Language property (Russian is default).

Ponkie, 2019-12-17:
Apologies, I am using the Saraff.Twain.NET with HTML samples that was on the website.

Thanks for the insight, I'm guessing that the problem was the driver itself.

Btw, on what function exactly do I change the Twain32.Language? Sorry for the questions :)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-17:
Try in Saraff.Twain.Aux_x86 (x64) in Program.cs
```c#
using(var _twain32=new Twain32()) {
    _twain32.Language = TwLanguage.ENGLISH;
    // ...
```

Ponkie, 2019-12-18:
Thanks for your help!

closed 2019-12-18 by Ponkie