saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #58: no response after fire XferDone event

wxh0800, 2019-12-18:
I have a Canon DR-M140 scanner, Saraff.Twain.NET works in most of scenes, But when I choose the highest resolution 600dpi and color mode is RGB to scan, It does not work, I debug into the code, normally, after XferDone event got fired, EndXfer event will get fired, but for 600dpi and RGB, only XferDone got fired, no EndXfer event got fired, and not image data gets back.

and in code TwainExternalProcess.cs

```			private void _OnEvent()
			{
				// var wholeOut = this._proc.StandardOutput.ReadToEnd();
				for (string _event = null; this.IsValid && (this.IsValid = !string.IsNullOrEmpty(_event = this._proc.StandardOutput.ReadLine()));)
				{
					var _res = TwainCommand.FromArray(Convert.FromBase64String(_event)) as EventHandlerTwainCommand;
					if (_res != null && this.FireEvent != null)
```

_event = this._proc.StandardOutput.ReadLine() read out {A1EF8BC7-9016-4FF1-962A-8F2C9726342C} (EndEvent Tag) instead of base64 code, so it raises an exception in
var _res = TwainCommand.FromArray(Convert.FromBase64String(_event)) as EventHandlerTwainCommand;

Could you help me to figure out what happens?

Thanks a lot

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-18:
Hello, @wxh0800 
Try check your scanner (for highest resolution 600dpi) using Saraff.Twain.Sample2 from [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

wxh0800, 2019-12-19:
I try to run  **Saraff.Twain.Sample2**, as it is based on **.NET 2.0** ( and it's reference **Saraff.Twain**), so import **Saraff.Twain** project to  **Saraff.Twain cs samples** solution, make it based on .net 4.6.1, I can't find .net 2.0 download url. 
when start **Saraff.Twain.Sample2**, it throws a "cann't load DSM", I debug into it, 

in Twain32.cs line146
```
this._hTwainDll = _Platform.Load(File.Exists(_twainDsm) && this.IsTwain2Enable ? _twainDsm : Path.ChangeExtension(Path.Combine(Environment.SystemDirectory, "..\\twain_32"), ".dll"));
```
**this._hTwainDll** return 0, and the 

But I see the combined file path "c:\\Windows\\System32\\..\\twain_32.dll" is correct.

In the project of my first comment, I can also see this exception in  Application event log. But it does not affect the scanning.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-19:
Hello, @wxh0800 
Ok, back to the first issue. Likely that issue related with the OutOfMemoryException, because image size is too big. Try use the Memory Transfer mode or File Transfer mode.

wxh0800, 2019-12-19:
Thanks a lot at first, but can you be kind to show me how to do this on  the Memory Transfer mode or File Transfer mode. I am new for this kind of staff. Because it is important for me, the requirement asks the continually multiple page scan.

Thank you very much

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-19:
See wiki pages [Buffered Memory Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2430.html) and [Disk File Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2420.html). 
Also you can see code of the Saraff.Twain.Sample2 and [Saraff.Twain.NET Outproc Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/outproc.html)

About the target which you can begin with, try see wiki page [Saraff.Twain.Aux](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainaux/)

closed 2019-12-24 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A