saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #60: .Net Core 3.0/1 support

makingbloke, 2019-12-31:
Hi Sarraff,

Is there any plans to support .Net Core 3.0 / 3.1? (I just tried to use the latest Nuget package with a .Net Core 3.1 WPF project and got a message stating it was compatible with an earlier version of the runtime so may not work with .Core 3)

Thanks

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2019-12-31:
Hello, @makingbloke 
Try using the [Saraff.TwainX.NET](https://www.nuget.org/packages/Saraff.TwainX.NET/), it's supported .NET Standard 2.0
Also, you can see the [Saraff.TwainX.NET](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainx/) wiki pages

closed 2020-01-05 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

John0King, 2020-03-21:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A  why create a new `Saraff.TwainX` instead of multi target ?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-03-22:
Hello, @John0King 
Because the Saraff.TwainX have different API.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-09-14:
Hello @makingbloke 
The [Saraff.Twain 1.0.32.732](https://www.nuget.org/packages/Saraff.Twain/1.0.32.732#supportedframeworks-body-tab) support .NET 6.0 (.net core 6.0)