saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #62: Enumeration FromRange(Range value) not returning the last value

khoke1, 2020-01-10:
Hello!  
Thank you first, for such great code!

I think I located a bug in **Twain32.Enumeration FromRange(Range value)**

I am working with 2 drivers: TypeGeneric Network TWAIN Driver Ver.5.67.01 & Canon Color Network ScanGear 2 Version 2.20.  Both drivers use "Range" for their XResolution and YResolution capabilities.  For both drivers, the resolution's maximum value is not returned in the enumeration created with the "FromRange" function.

For example, the Canon's driver allows resolutions from 75 to 600 in a step size of 1.  The number of items created in the object is 525: (Max value 600 - min value 75)/ (Step size 1). 
But the actual total number of values counting 75 through 600 is technically **526** different resolutions.

To resolve my issue, I changed the line of code:

object[] _items = new object[(int)((Convert.ToSingle(value.MaxValue) - Convert.ToSingle(value.MinValue)) / Convert.ToSingle(value.StepSize))];

object[] _items = new object[(int)(((Convert.ToSingle(value.MaxValue) - Convert.ToSingle(value.MinValue)) **+ 1)** / Convert.ToSingle(value.StepSize))

My code now returns the max value in the array for all drivers using range for X & Y resolution.

Thoughts? Thanks!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-01-13:
Hello, @khoke1 
You are probably right. I will make the appropriate changes in the next release.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-01-18:
Hello, @khoke1 
See 292288bd4032ce867faa5044838c8659babb3042

closed 2020-01-24 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A