saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #66: Cannot run Service Sample

khoke1, 2020-01-28:
Hello!  I am trying to run the service sample 1, but continue to get an exception, crashing the application.  I have the service running.

Thanks!
Kelly

**Message**: 
"Win32Exception: CreateProcessAsUser failed.\r\n   at Saraff.Twain.Aux.ImpersonationUtils._LaunchProcessAsUser(String fileName, IntPtr token, IntPtr envBlock, IntPtr& hStdIn, IntPtr& hStdOut, IntPtr& hStdError)\r\n   at Saraff.Twain.Aux.ImpersonationUtils.RunAsCurrentUser(String fileName)\r\n   at Saraff.Twain.Aux.TwainExternalProcess.AuxProcess.CreateProcess(String fileName)\r\n   at Saraff.Twain.Aux.TwainExternalProcess.Execute(String fileName, Action`1 execCallback)\r\n   at Saraff.Twain.Service.TwainService.GetSources() in C:\\~Code\\Saraff.Twain.ServiceSamples\\sources\\Saraff.Twain.Service\\TwainService.svc.cs:line 56\r\n\r\n"
**StackTrace**:
"   at System.Runtime.Remoting.Proxies.RealProxy.HandleReturnMessage(IMessage reqMsg, IMessage retMsg)\r\n   at System.Runtime.Remoting.Proxies.RealProxy.PrivateInvoke(MessageData& msgData, Int32 type)\r\n   at Saraff.Twain.ServiceSample1.TwainServiceReference.ITwainService.GetSources()\r\n   at Saraff.Twain.ServiceSample1.TwainServiceReference.TwainServiceClient.GetSources() in C:\\~Code\\Saraff.Twain.ServiceSamples\\sources\\Saraff.Twain.ServiceSample1\\Service References\\TwainServiceReference\\Reference.cs:line 824\r\n   at Saraff.Twain.ServiceSample1.Form1._Load() in C:\\~Code\\Saraff.Twain.ServiceSamples\\sources\\Saraff.Twain.ServiceSample1\\Form1.cs:line 63\r\n   at Saraff.Twain.ServiceSample1.Form1.OnLoad(EventArgs e) in C:\\~Code\\Saraff.Twain.ServiceSamples\\sources\\Saraff.Twain.ServiceSample1\\Form1.cs:line 56"

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-01-29:
Hello, @khoke1 
Try reinstall the Saraff.Twain.Service using Saraff.Twain.Service_x64.msi (or Saraff.Twain.Service_x86.msi) from the Saraff.Twain.NET Service Samples

khoke1, 2020-01-29:
Not sure what is happening, I was unable to uninstall the service - with this error.

I am unable to use the installers, since the service is already installed.  So, I tried to uninstall with installutil /u and get these errors:

An exception occurred while trying to find the installers in the C:\~Code\Saraff.Twain.ServiceSamples\sources\Saraff.Twain.Service\Saraff.Twain.Service.exe assembly.
System.Reflection.ReflectionTypeLoadException: Unable to load one or more of the requested types. Retrieve the LoaderExceptions property for more information.
Aborting installation for C:\~Code\Saraff.Twain.ServiceSamples\sources\Saraff.Twain.Service\Saraff.Twain.Service.exe.
|

khoke1, 2020-01-29:
Close due to resolution.

closed 2020-01-29 by khoke1