saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #69: NuGet Packege Version

Esam-Bustaty, 2020-06-03:
Hi,

It seems that the nuget package has the version 678 (for Saraff.Twain, 693 for TwainX).

Can you please upload the latest version to NuGet repo.

Thanks,

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-06-03:
Hello, @Esam-Bustaty 

The [Saraff.Twain.NET 1.0.29.678](https://www.nuget.org/packages/Saraff.Twain/1.0.29.678) (see b15cce27b71218a8765526d690ba0e986f9b120b commit)  corresponds the [Saraff.TwainX.NET 1.0.29.693](https://www.nuget.org/packages/Saraff.TwainX.NET/1.0.29.693) (see 344050d5dbbf5bda08089eb6b58d169f36028df6 commit)

The difference in revision numbers in package versions is determined by the presence of separate code branches for each package

Esam-Bustaty, 2020-06-03:
Thanks for the reply

ok, so let me ask a different way: the commit 292288bd4032ce867faa5044838c8659babb3042 solves a problem that's still unsolved in the nuget package.

My scanner supports resolutions upto 600 dpi, but when I list the values returned by twain.Capabilities.Resolutions.Ger() I get only up to 599 dpi.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-06-04:
Hello, @Esam-Bustaty 
uploaded.

Esam-Bustaty, 2020-06-04:
Thanks for your efforts, Have good day :)

closed 2020-06-04 by Esam-Bustaty