saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #71: Cann't load dsm

gigimignon, 2020-06-12:
With Twain32, i have no problem but with .net core version, i have the message "Cann't load dsm" 

With 32 bits process.

A difference existe between Twain32.cs and TwainX.cs. 32 bit dll not test for TwainX.cs in source code.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-06-12:
Hello, @gigimignon 
The Twain32 can use TWAIN 1.x (twain_32.dll) and TWAIN 2.x (TWAINDSM.DLL), but TwainX use only TWAIN 2.x (TWAINDSM.DLL). So try install the TWAINDSM.DLL (see [Installation of the Data Source Manager](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w0200.html)

gigimignon, 2020-06-12:
Thanks
A last question, why don't use twain32.dll?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-06-12:
Because twain_32.dll use the win32 callback function for conversation between a client app and a data source. but the dotnet core don't support win32 (and winforms)

closed 2020-06-12 by gigimignon