saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #74: Scanner settings

jueqing1015, 2020-07-25:
I need to perform automatic scanning now. Before automatic scanning, I need to pop up the scanner's own interface for parameter setting. How do I do this?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-07-25:
Hello, @jueqing1015 
See the Saraff.Twain.Sample2 from the [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html).

Also you can see wiki about the [Saraff.Twain.NET](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/)

jueqing1015, 2020-07-25:
Hello, the setting method in example 2 is to set through the software instead of using the scanner window to set

jueqing1015, 2020-07-25:
Maybe my description is not too clear
This is my scan comes with the scan interface
[image]
This is the setting interface that comes with my scan
[image]

I now want to bring up the scan built-in settings interface

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-07-26:
Hello, @jueqing1015
See the Saraff.Twain.Sample**1** from the [Saraff.Twain.NET CS Samples.](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)
Also you can see [wiki](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/) about the Saraff.Twain.NET ([Minimal implementation](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w1000.html) & [Alternative User Interfaces](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2300.html))

jueqing1015, 2020-07-30:
I checked and found that it does not provide the kind of functional interface I mentioned. I have implemented it through other methods. You can close this question, thank you

closed 2020-07-30 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A