saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #75: Access Violation error

saurabhabh, 2020-09-14:
This is the code that I have written 
```

using System;
using System.Collections.Generic;
using System.Collections.ObjectModel;
using System.Drawing;
using System.Linq;
using System.Runtime.InteropServices;
using System.Text;
using System.Threading.Tasks;
using System.Windows;
using System.Windows.Data;
using System.Windows.Media.Imaging;
using System.Windows.Threading;
using Saraff.Twain;
using Saraff.Twain.Aux;

namespace Inweon_Grain_Management_System.Services
{
    class TWAINScannerService
    {
        Bitmap scanImage = null;
        Saraff.Twain.Twain32 twain = null;
        public TWAINScannerService()
        {
            twain = new Saraff.Twain.Twain32();
            twain.ShowUI = false;
            twain.AppProductName = "Saraff.Twain";
            twain.IsTwain2Enable = true;
            twain.OpenDSM();
            twain.SetupFileXferEvent += new System.EventHandler<Saraff.Twain.Twain32.SetupFileXferEventArgs>(this._twain32_SetupFileXferEvent);
            twain.FileXferEvent += new System.EventHandler<Saraff.Twain.Twain32.FileXferEventArgs>(this._twain32_FileXferEvent);

            twain.AcquireError += new System.EventHandler<Saraff.Twain.Twain32.AcquireErrorEventArgs>(this._twain32_AcquireError);
          
        }
        public void Scan()
        {
            #region Set Capabilities

            twain.SourceIndex = 0;
            twain.OpenDataSource();

            try
            {
                twain.SetCap(TwCap.XResolution, 600);
            }
            catch
            {
            }

            try
            {
                twain.SetCap(TwCap.YResolution, 600);
            }
            catch
            {
            }

            try
            {
                twain.SetCap(TwCap.IPixelType, TwPixelType.RGB);
            }
            catch
            {
            }

            try
            {
                twain.SetCap(TwCap.IXferMech, TwSX.File);
            }
            catch
            {
            }

            try
            {
                twain.SetCap(TwCap.ImageFileFormat, TwFF.Bmp);
            }
            catch
            {
            }

            try
            {
                twain.Capabilities.Indicators.Set(false);
            }
            catch
            {
            }

            twain.Capabilities.XferCount.Set(1);

            #endregion

            twain.Acquire();

        }

      
        private void _twain32_AcquireError(object sender, Twain32.AcquireErrorEventArgs e)
        {
            throw e.Exception;
        }
        private void _twain32_SetupFileXferEvent(object sender, Twain32.SetupFileXferEventArgs e)
        {
            try
            {
                e.FileName = string.Format(@"FileXferTransfer_{0}.{1}", DateTime.Now.ToString("yyyyMMddHHmmss"), this.twain.Capabilities.ImageFileFormat.GetCurrent().ToString().ToLower());
            }
            catch (Exception ex)
            {
                throw ex;
            }
        }

        private void _twain32_FileXferEvent(object sender, Twain32.FileXferEventArgs e)
        {
            try
            {
                this.scanImage = Image.FromFile(e.ImageFileXfer.FileName) as Bitmap;
            }
            catch (Exception ex)
            {
                throw ex;
            }
        }
    }
}

```

Now When **Acquire()** is called it gives me below error with attached stack trace 
**_Attempted to read or write protected memory. This is often an indication that other memory is corrupt._**
>    at Saraff.Twain.Twain32._DsmEntry.DsInvoke[T](TwIdentity origin, TwIdentity dest, TwDG dg, TwDAT dat, TwMSG msg, T& data)
>    at Saraff.Twain.Twain32._EnableDataSource()
>    at Saraff.Twain.Twain32.Acquire()
>    at Inweon_Grain_Management_System.Services.TWAINScannerService.Scan() in E:\Saurabh\Work\Inweon\inweon-grain-management-system\Inweon Grain Management System\Services\TWAINScannerService.cs:line 96
>    at Inweon_Grain_Management_System.Services.MockMeasurementService.LoadGrainReadings(GateKeeperService gateKeeperService) in E:\Saurabh\Work\Inweon\inweon-grain-management-system\Inweon Grain Management System\Services\MockMeasurementService.cs:line 23
>    at Inweon_Grain_Management_System.MainWindowUSA.Measure(GateKeeperService gateKeeperService) in E:\Saurabh\Work\Inweon\inweon-grain-management-system\Inweon Grain Management System\MainWindowUSA.xaml.cs:line 216
>    at Inweon_Grain_Management_System.MainWindowUSA.<RunTask>b__25_0() in E:\Saurabh\Work\Inweon\inweon-grain-management-system\Inweon Grain Management System\MainWindowUSA.xaml.cs:line 194
>    at System.Threading.Tasks.Task.Execute()
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>    at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>    at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
>    at System.Threading.Tasks.Task.ExecuteEntry(Boolean bPreventDoubleExecution)
>    at System.Threading.ThreadPoolWorkQueue.Dispatch()

Any help will be really great. I am running a x64 machine and have installed a virtual scanner. (https://developer.dynamsoft.com/dwt/kb/2659)

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-09-15:
Hello, @saurabhabh 
I think that it related with the multithreading accessing. If you need to the async invocation that see the [Saraff.Twain.NET Outproc Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/outproc.html) or the [Saraff.Twain.NET Service Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/service.html)

saurabhabh, 2020-09-15:
I followed one of the WPF examples (sample2) and it worked. Thanks for the quick response though.

closed 2020-09-21 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A