saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #77: The error of _twain32 after setting ShowInTaskbar on C# Winform

OilPrince, 2020-10-16:
Hello!

Before we get to the point, I would like to thank you for opening up a good solution and for your kind response on the issue bulletin board.

I am producing a scan program with C# Winform using Saraff TWAIN.

If you close the C# Winform of the program and make it appear again, Saraff TWAIN is experiencing a problem.

Here's how to close C# Winform:
This.ShowInTaskbar = false;

this.visible = false;
this.Hide();

Here's how to reopen C# Winform:
this.Show();
this.visible = true;

if (this.WindowState == FormWindowState.Minimized)
{
This.WindowState = FormWindowState.Normal;
}

This.ShowInTaskbar = true;

The following error occurs when scanning or selecting the scanner (_twain32.SelectSource();)) after re-opening as shown above:
DG DAT MSG out of expected sequence.

This after the program runs.After setting ShowInTaskbar, we have confirmed that the method in _twain32 has been experiencing problems since then.

A way to get back, ie this.There is a way not to set ShowInTaskbar, but a fundamental solution is needed.

Please give me some advice on this!

Thanks!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-10-16:
Hello, @OilPrince 
In first, thanks for your feedback.
About the issue, this problem related with your a Data Source (scanner driver). Likely, it didn't expect that you to hide the window when the driver was running. Or your invoke the Twain32 methods in the wrong sequence. 
For more information see the Controlling a TWAIN Session from Your Application (TWAIN 2.4 Specification, 3-9) in the [TWAIN 2.4 Specification](https://twain.org/specification/)
Also, you can see the Saraff.Twain.Sample2 in the [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

OilPrince, 2020-10-17:
Hi, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A!
Nice to meet you!

Thank you for your quick and kind reply.

I checked my program again as you told me.

And we tested it with the example application(Saraff.Twain.Sample2) you provided.
First scanning works normally on the example application. By the way, I set ShowInTaskbar in C# Winform(This.ShowInTaskbar = false;) and then there's no response even when I scan it.

When you change the dataSourcesToolStripComboBox and scan it, the results were the same.

The tested program can be downloaded from the following URL:

http://naver.me/Gom4KOfL

I want to know if it's an error in the Saraff TWAIN program.
If you say it's an error, please correct it!

Thank you again for your kindness.

Goodbye!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-10-17:
Hello, @OilPrince 
In this case, try set the Twain32.Parent to the null (you can do it in the vs designer). Because if the ShowInTaskbar change to the false that it release internal win32 handle. And the Data Source will not be able to send a events (triplets) into the application.
If set the Twain32.Parent to the null that the Twain32 will be use internal the window for event processing.

OilPrince, 2020-10-18:
Hello, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A!

Thank you for your kind and detailed reply.

I retested the example application by modifying it according to your guide.
As a result, the application scanning function worked well without errors.

It has been confirmed that no problem occurs if the set part(This._twain32.Parent = this;) is removed at all or the syntax is modified(this._twain32.Parent = null;).

I'm very glad that the problem has been solved.

Thank you very much for your kindness.

I hope you will develop the Saraff solution into a better solution as you have done so far.

I wish you good luck all the time.

Goodbye!

PS : If there are no more comments, I will close this issue in a few days.

closed 2020-10-23 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A