saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #78: AutoDiscardBlankPages capability is not working

rohitnrynn, 2020-10-19:
Hi,
    We tried to set AutoDiscardBlankPages to TwBP.Auto only if the scanner has the capability. The value is getting set correctly. But while acquiring, blank pages are also getting scanned. We are using Canon Scanners. The code we wrote is given below. 

```
if ((_twain32.IsCapSupported(TwCap.AutoDiscardBlankPages) & TwQC.Set) != 0)
{
    _twain32.SetCap(TwCap.AutoDiscardBlankPages, TwBP.Auto);
}
```

On our search, we found an issue with Canon scanner. Please see the link
[](http://www.atalasoft.com/KB2/ExportPDF50118.aspx)

Can you please guide us to make this capability work.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-10-20:
Hello, @rohitnrynn 
This the issue related directly with your scanner. You may use the solution from the "FIX: AutoDiscardBlankPages Doesn't Work With Canon Scanners". See follow links: 
- [General (not strongly typed) accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2120.html)
- [Capability Containers](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2121.html)
- [Custom capability](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2122.html)

rohitnrynn, 2020-10-23:
Ok will check and come back.

closed 2020-10-29 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A