saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #80: Пробую запустить на базе Net Core 3 чтобы впоследствии юзать на MacOs

zabralex85, 2020-10-30:
Так и должно быть?

При этом в одной сборке правда под Net 4 старой работало и то и то.
[image]
[WPF-Demo.exe.zip](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/files/5465585/WPF-Demo.exe.zip)

Libs:

```
<PackageReference Include="Saraff.TwainX.Extensions" Version="1.0.2.616" />
<PackageReference Include="Saraff.TwainX.NET" Version="1.0.30.696" />
```

Virtual Scanners:

```
FreeImage Software Scanner
ScanWorks TWAIN Importer
```

CODE:

```
            var twainX = new TwainX { ShowUI = false };

            twainX.OpenDSM();

            for (var i = 0; i < twainX.SourcesCount; i++)
            {
                Console.WriteLine("{0}: {1}{2}", i, twainX.GetSourceProductName(i), twainX.GetIsSourceTwain2Compatible(i) ? " (TWAIN 2.x)" : string.Empty);
            }

            twainX.OpenDSM();

            bool res = twainX.SelectSource();
            var source = twainX.GetDefaultSource();

            try
            {
                twainX.OpenDataSource();
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);
                Console.Read();
                return;
            }
```

RESULT:
```
Qt: Untested Windows version 6.2 detected!
0: TWAIN2 FreeImage Software Scanner (TWAIN 2.x)
1: ScanWorks TWAIN Importer
DS don't supported TWAIN 2.x specification.
```

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-10-31:
Добрый день, @zabralex85 

> Если программу запускать x86 то видно 2 сканнера, x64/any - только FreeImage Software Scanner
> Так и должно быть?

Ну, как бы, да. 

Вероятно, дело в том, что вы используете пакет [Saraff.TwainX.NET](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwainx/), который поддерживает только спецификацию TWAIN 2.х.
А в примере, на который вы ссылаетесь, используется пакет [Saraff.Twain.NET](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/), который поддерживает и TWAIN 2.х и TWAIN 1.х

Ко всему прочему (и это главная причина), вы убрали из кода OutProc взаимодействие, поэтому теперь, в зависимости от разрядности образа  исполняемого модуля (сборки), видны или 32-разрядные или 64-разрядные источники (64-разрядные источники (DS) не могут быть загружены в 32-разрядные программы; и наоборот, т.к., по умолчанию используется InProc взаимодействие с DS).

Как-то так.

zabralex85, 2020-10-31:
при этом как понимаю TwainX (twain 2 only) - core совместим (net standard), а TwainNet (twain 1/2) - нет?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2020-11-02:
Добрый день, @zabralex85

> при этом как понимаю TwainX (twain 2 only) - core совместим (net standard), а TwainNet (twain 1/2) - нет?

Так и есть.

closed 2020-11-07 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A