saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #84: Attempted to read or write protected memory. This is often an indication that other memory is corrupt

sishen2, 2021-07-30:
use example "Saraff.TwainX.NETCoreSample2"  Set 72 dpi  throw Exception "Attempted to read or write protected memory. This is often an indication that other memory is corrupt" ,

[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-07-31:
Hello, @sishen2 
Likely it related with your scanner. For more info you need show the stack trace of the exception. Also you can try different examples with same params.

sishen2, 2021-08-03:
But I won't have a problem using the ntwain example 
https://github.com/nozerowu/ntwain/tree/master/src/NTwain

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-08-03:
Hello, @sishen2
For more info you need show the stack trace of the exception. Also you can try different examples ( https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/#samples ) with same params.

closed 2021-08-08 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A