saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #85: Multiple images for one scan

chiponline89, 2021-08-31:
Hi,
Currently, I have an issue when using scanner "HP scanjet Enterprise Flow 7000 s3" to scan. 

Multiple images output for one scan. I receipted 2  same images for 1 page. And 4 images (2 same images of first page and 2 same images of back page) if duplex scan. 

Could you help me about this issue?

Thank you so much!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-09-03:
Hello, @chiponline89 
Perhaps this is due to the fact that you are subscribing twice to the event of receiving the image.

chiponline89, 2021-09-04:
> Hello, @chiponline89
> Perhaps this is due to the fact that you are subscribing twice to the event of receiving the image.

closed 2021-09-04 by chiponline89

chiponline89, 2021-09-04:
Hello, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

- I also dowloaded example in this link : https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/
-> Run it -> I got same issue in my computer

- After, I run one more time with this code in other computer -> I not got this issue.

=> I think that maybe the main reason is driver of scanner
- I will try uninstall this driver and reinstall it